xmikos/qspectrumanalyzer #1: Install instructions

andresv, 2015-05-06:
Hi,

I am trying to build it under Ubuntu 14.10. It is quite a pain to meet all the python3 dependencies without knowing exactly what is needed and how they should be installed.

It would be a good reference for other platforms as well if for example there is proper install instructions for Ubuntu.

andresv, 2015-05-06:
https://github.com/xmikos/qspectrumanalyzer/pull/2

xmikos, 2015-05-07:
Hello,

please look at my comment under your pull request. I can add install instructions for Ubuntu into README, but I would prefer if someone can create proper Debian/Ubuntu source package (`control` file or what Debian/Ubuntu uses).

xmikos, 2015-12-10:
I have added installation instructions to README.rst, this can be closed.

closed 2015-12-10 by xmikos