xmikos/qspectrumanalyzer #4: Support for sdr_power

reldred, 2016-05-23:
sdr_power is a modified rtl_power that supports interfacing with a couple more radio types, BladeRF and HackRF at the moment, and seems to still be under active development. From initial testing it appears to provide a data stream in the same format as rtl_power. Most command line flags are identical to rtl_power, however a new -r flag is provided for selecting the radio type and specific radio settings which has prevented using it as an immediate drop-in replacement for rtl_power.

[https://github.com/polygon/sdr-power](https://github.com/polygon/sdr-power)

This strikes me as the easiest way of letting QSpectrumAnalyzer support the BladeRF/HackRF and frequencies up to 6Ghz.

Thanks!

xmikos, 2016-05-27:
Hi,

thank you for info. I have recently started to work on SoapySDR based backend which should support just about any SDR hardware out there (sdr_power seems to support only RTL-SDR and HackRF for now). Also it should  be much more flexible (similar to rtl_power_fftw).

I should get Airspy and SDRplay dongles soon, so it will be tested with them.

reldred, 2016-05-28:
Excellent to hear, thank you.

I have a HackRF arriving this week which I can also test with.

mightytarzan, 2017-01-09:
Quick questions, any news on the SoapySDR backend front?

xmikos, 2017-02-14:
I have now finally started to work on QSpectrumAnalyzer again, sorry that it took that long (family, new job, etc.). I have already merged *hackrf_sweep* backend (by @miek) and added preliminary support for my own *soapy_power* backend (which I will publish on GitHub soon). Also [rx_power](https://github.com/rxseger/rx_tools) will be supported (I have already added backend for it, but didn't test it yet).

xmikos, 2017-03-11:
Now soapy_sdr backend is finally stable and latest version of QSpectrumAnalyzer with soapy_sdr backend works great with RTL-SDR, HackRF and Airspy. There are still some issues with LimeSDR (will fix it soon) and SDRPlay (don't know how long will it take, latest libsdrplay / soapysdrplay-git seems broken in Arch Linux, I will need to first find where is the problem).

closed 2017-03-11 by xmikos