xmikos/qspectrumanalyzer #7: FileNotFoundError

stevevaius77, 2017-02-12:
Nothing on running for HackRF One (firmware 2017.02.1)
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/qspectrumanalyzer/backend.py", line 49, in run
    self.process_start()
  File "/usr/local/lib/python3.5/dist-packages/qspectrumanalyzer/backend.py", line 214, in process_start
    universal_newlines=True)
  File "/usr/lib/python3.5/subprocess.py", line 676, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.5/subprocess.py", line 1282, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'rtl_power_fftw'
rtl_power_fftw params:
{'bin_size': 10.0,
 'bins': 2000,
 'crop': 0.0,
 'freq_range': 21000000.0,
 'gain': 0,
 'hops': 2,
 'interval': 10.0,
 'min_overhang': 0.0,
 'overhang': 19000000.0,
 'overlap': 0.0,
 'ppm': 0,
 'sample_rate': 20000000,
 'single_shot': False,
 'start_freq': 87.0,
 'stop_freq': 108.0,
 'time': 5.0}

reldred, 2017-02-12:
Try these forks; https://github.com/miek/qspectrumanalyzer / https://github.com/mossmann/qspectrumanalyzer

Also, you've also got the backend set as rtl_power_fftw which is for RTL dongles only, HackRF support is in miek's fork using hackrf_sweep, it's all set in the settings window. Michael Ossmann has a few more tweaks in his fork as well. I imagine they'll all be going upstream back to here at some point.

stevevaius77, 2017-02-13:
Thanks! I will try and let you know

Rustu

xmikos, 2017-02-14:
hackrf_sweep fork as been merged, you can now use master branch here.

closed 2017-02-14 by xmikos