xmikos/qspectrumanalyzer #8: No "File" and "Help" menu on top

stevevaius77, 2017-02-13:
[image]
There is no file and help menu on top to change device. And on Terminal again no supported device. What should be done to get Hackrf One to work with it? (Firmware 2017.02.1)
[image]

xmikos, 2017-02-14:
Sorry, I couldn't reproduce this issue. Isn't Ubuntu somehow messing with top menus of Qt 4 applications?

Montilien, 2017-02-21:
Hi,
I have exactly the same issue...hackrf_info and hackrf_sweep checked works fine...
André

xmikos, 2017-02-21:
Do you also use Ubuntu? I didn't have time to test in on latest Ubuntu yet, but will try...

stevevaius77, 2017-02-21:
I solved it by reinstalling Ubuntu freshly and installing first all hackrf.git and reflashing firmware of my hackrf to be sure. I used this git https://github.com/miek/qspectrumanalyzer and install it not by system wide, but locally (pip3 install --user qspectrumanalyzer) which is now works smoothly

Montilien, 2017-02-21:
Ho,
Yes i have a fresh Ubuntu upgrade to 16.04 release.
Regards

Montilien, 2017-02-21:
> According to my findings, there is option in Ubuntu: Settings / Appearance / Behavior. Check the options for show the menu in the window's title bar and for always show menus.
>
> Try it and let me know (I didn't tested it, this is what I have found on Google).

Congratulations...you are the best...
I have changed the system settings And all works fine...
Thanks you vert much
André

xmikos, 2017-02-22:
I have now tried it in Ubuntu 16.10 (in VirtualBox) and you don't have to change the settings. Menu is actually there, it is just pulled out of QSpectrumAnalyzer window and shown in the left side of the main Ubuntu to top bar (after you move mouse pointer there). I guess this is standard behavior of all windows in Ubuntu Desktop. See attached screenshot:

[image]

closed 2017-03-11 by xmikos