xmikos/qspectrumanalyzer #13: Trying to run on Ubuntu with direct cloned version

sm3ulc, 2017-02-15:
I use Ubuntu 16.04.2 LTS and do the following:

git clone https://github.com/xmikos/qspectrumanalyzer
cd  qspectrumanalyzer
python setup.py install

When I try to run I get:

qspectrumanalyzer
Traceback (most recent call last):
  File "/usr/local/bin/qspectrumanalyzer", line 9, in <module>
    load_entry_point('QSpectrumAnalyzer==1.4.0', 'gui_scripts', 'qspectrumanalyzer')()
  File "build/bdist.linux-x86_64/egg/qspectrumanalyzer/__main__.py", line 557, in main
  File "build/bdist.linux-x86_64/egg/qspectrumanalyzer/__main__.py", line 158, in __init__
TypeError: super() takes at least 1 argument (0 given)

Have I missed some step or error in code?

xmikos, 2017-02-15:
You have to use Python 3, this seems like you are running it with old Python 2.

Btw. you should not install applications system-wide outside of your package manager. This way you will mess up your Linux system. You can use *pip* and install it only for your local user (as described in README):

    sudo apt-get install python3-pip python3-pyqt4 python3-numpy
    cd qspectrumanalyzer/
    pip3 install --user .

In this case qspectrumanalyzer executable will be placed in `~/.local/bin` directory, you can add it to your PATH in `~/.bashrc`

sm3ulc, 2017-02-16:
Thank you for great solution! Got it working with the hackrf-one.

Kind regards,

// David

closed 2017-02-16 by sm3ulc

indrajitbarve, 2018-01-03:
Hello xmikos, 

after installing the above-mentioned command still, it's trying to run using python2.7 . How to change the paths.

indrajit@indrajit-deepti:~/radiognu/qspectrumanalyzer-master$ sudo pip3 install --user .
Unpacking /home/indrajit/radiognu/qspectrumanalyzer-master
  Running setup.py (path:/tmp/pip-zkmq8chy-build/setup.py) egg_info for package from file:///home/indrajit/radiognu/qspectrumanalyzer-master
    
Requirement already satisfied (use --upgrade to upgrade): soapy-power>=1.6.0 in /home/indrajit/.local/lib/python3.4/site-packages (from QSpectrumAnalyzer==2.2.0)
Requirement already satisfied (use --upgrade to upgrade): pyqtgraph>=0.10.0 in /home/indrajit/.local/lib/python3.4/site-packages (from QSpectrumAnalyzer==2.2.0)
Requirement already satisfied (use --upgrade to upgrade): Qt.py in /home/indrajit/.local/lib/python3.4/site-packages (from QSpectrumAnalyzer==2.2.0)
Requirement already satisfied (use --upgrade to upgrade): numpy in /usr/lib/python3/dist-packages (from soapy-power>=1.6.0->QSpectrumAnalyzer==2.2.0)
Requirement already satisfied (use --upgrade to upgrade): simplesoapy>=1.5.0 in /home/indrajit/.local/lib/python3.4/site-packages (from soapy-power>=1.6.0->QSpectrumAnalyzer==2.2.0)
Requirement already satisfied (use --upgrade to upgrade): simplespectral in /home/indrajit/.local/lib/python3.4/site-packages (from soapy-power>=1.6.0->QSpectrumAnalyzer==2.2.0)
Installing collected packages: QSpectrumAnalyzer
  Found existing installation: QSpectrumAnalyzer 2.1.0
    Uninstalling QSpectrumAnalyzer:
      Successfully uninstalled QSpectrumAnalyzer
  Running setup.py install for QSpectrumAnalyzer
    
    Installing qspectrumanalyzer script to /home/indrajit/.local/bin
Successfully installed QSpectrumAnalyzer
Cleaning up...
indrajit@indrajit-deepti:~/radiognu/qspectrumanalyzer-master$ qspectrumanalyzerTraceback (most recent call last):
  File "/usr/local/bin/qspectrumanalyzer", line 9, in <module>
    load_entry_point('QSpectrumAnalyzer==2.2.0', 'gui_scripts', 'qspectrumanalyzer')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 351, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2363, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/usr/local/lib/python2.7/dist-packages/QSpectrumAnalyzer-2.2.0-py2.7.egg/qspectrumanalyzer/__main__.py", line 7, in <module>
    from qspectrumanalyzer import backends
  File "/usr/local/lib/python2.7/dist-packages/QSpectrumAnalyzer-2.2.0-py2.7.egg/qspectrumanalyzer/backends/__init__.py", line 5, in <module>
    from qspectrumanalyzer import subprocess
  File "/usr/local/lib/python2.7/dist-packages/QSpectrumAnalyzer-2.2.0-py2.7.egg/qspectrumanalyzer/subprocess.py", line 57
    def __init__(self, *pargs, console=True, **kwargs):
                                     ^
SyntaxError: invalid syntax