xmikos/qspectrumanalyzer #15: HackRF: No supported devices found

jimport, 2017-02-18:
I'm using qspectrumanalyzer version 1.4.0 on Arch Linux. When I hit the Start button, this gets displayed in the console:
```
rtl_power params:
{'bin_size': 10.0,
 'crop': 0.0,
 'gain': 0,
 'hops': 0,
 'interval': 10.0,
 'ppm': 0,
 'single_shot': False,
 'start_freq': 96.0,
 'stop_freq': 122.0}

Number of frequency hops: 10
Dongle bandwidth: 2600000Hz
Downsampling by: 1x
Cropping by: 0.00%
Total FFT bins: 5120
Logged FFT bins: 5120
FFT bin size: 5078.12Hz
Buffer size: 16384 bytes (3.15ms)
Reporting every 10 seconds
No supported devices found.
```
The HackRF is flashed with the latest firmware version 2017.02.1 and latest host software. The `hackrf_sweep` command is working well.

Any suggestions? Thanks in advance!

xmikos, 2017-02-18:
Support for HackRF is only in git master, there hasn't been new release yet. You must install it from git and select hackrf_sweep backend in settings.

jimport, 2017-02-18:
Thanks! But hackrf_sweep is not in the Backend list, when I install it with `makepkg -sri --skipchecksums`.

When I install it with `pip3 install --user .` as suggested in #13 I get
```
Traceback (most recent call last): 
  File "/usr/bin/qspectrumanalyzer", line 11, in <module>
    load_entry_point('QSpectrumAnalyzer==1.4.0', 'gui_scripts', 'qspectrumanalyzer')()
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 560, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2648, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2302, in load
    return self.resolve()
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2308, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
 File "/home/jim/.local/lib/python3.6/site-packages/qspectrumanalyzer/__main__.py", line 7, in <module>
    from qspectrumanalyzer import backends
ImportError: cannot import name 'backends'
```

xmikos, 2017-02-18:
Have you followed exactly this procedure?

```
git clone https://github.com/xmikos/qspectrumanalyzer.git
cd qspectrumanalyzer
pip3 install --user .
```

This definitely works on Arch Linux for me. Maybe you also have to uninstall previously installed qspectrumanalyzer 1.4.0 before this.

jimport, 2017-02-18:
Ok, I figured it out. According to the stacktrace the backends couldn't be loaded:
```
Traceback (most recent call last):
...
  File "/home/jim/.local/lib/python3.6/site-packages/qspectrumanalyzer/__main__.py", line 7, in <module>
    from qspectrumanalyzer import backends
ImportError: cannot import name 'backends'
```
So I had a look in `/home/jim/.local/lib/python3.6/site-packages/qspectrumanalyzer/` and the backends folder was missing. Copied it and it's working like a charme. 
Is there something wrong with the install script?

xmikos, 2017-02-18:
You were right, `qspectrumanalyzer.backends` package has been missing in `setup.py` script. I have fixed it now, could you please uninstall it (and check that `/home/jim/.local/lib/python3.6/site-packages/qspectrumanalyzer/` is really completely deleted) and then try that install procedure from https://github.com/xmikos/qspectrumanalyzer/issues/15#issuecomment-280881417 again? Just to be sure that it really works now...

Strange that it worked for me, maybe I had backends already installed from my previous tests.

jimport, 2017-02-18:
No it's not. The backends folder is not getting deleted.

xmikos, 2017-02-18:
Yes, it's pushed now (also fix of fix ;-)). If you have copyed backends directory manually, you have to also delete it manually (after pip3 uninstall delete whole `/home/jim/.local/lib/python3.6/site-packages/qspectrumanalyzer/`).

Then please try install procedure again (without manually copying backends directory) and tell me if it works.

jimport, 2017-02-19:
Thats it. Works perfect now! Thanks a lot!

xmikos, 2017-02-19:
Great, thanks for testing it.
Btw. I have now also created qspectrumanalyzer-git package in AUR: https://aur.archlinux.org/packages/qspectrumanalyzer-git/

closed 2017-02-19 by xmikos