xmikos/qspectrumanalyzer #17: Win10 (cygwin64) ****HDF5 library version mismatched error***

ipl2004, 2017-02-24:
Hi, relatively a novice, build and install went well with no errors, however I had and error and core dump on running.

Warning! ***HDF5 library version mismatched error***
The HDF5 header files used to compile this application do not match
the version used by the HDF5 library to which this application is linked.
Data corruption or segmentation faults may occur if the application continues.
This can happen when an application was compiled by one version of HDF5 but
linked with a different version of static or shared HDF5 library.
You should recompile the application or check your shared library related
settings such as 'LD_LIBRARY_PATH'.
You can, at your own risk, disable this warning by setting the environment
variable 'HDF5_DISABLE_VERSION_CHECK' to a value of '1'.
Setting it to 2 or higher will suppress the warning messages totally.
Headers are 1.8.15, library is 1.8.17
            SUMMARY OF THE HDF5 CONFIGURATION
            =================================

General Information:
-------------------
                   HDF5 Version: 1.8.17
                  Configured on: Sun May 15 08:43:52 CEST 2016
                  Configured by: marco@GE-MATZERI-EU
                 Configure mode: production
                    Host system: x86_64-unknown-cygwin
              Uname information: CYGWIN_NT-6.1 GE-MATZERI-EU 2.5.1(0.297/5/3) 2016-04-21 22:14 x86_64 Cygwin
                       Byte sex: little-endian
                      Libraries: shared
             Installation point: /usr

Compiling Options:
------------------
               Compilation Mode: production
                     C Compiler: /usr/bin/gcc
                         CFLAGS: -ggdb -O2 -pipe -Wimplicit-function-declaration -fdebug-prefix-map=/cygdrive/e/cyg_pub/devel/hdf5/hdf5-1.8.17-1.x86_64/build=/usr/src/debug/hdf5-1.8.17-1 -fdebug-prefix-map=/cygdrive/e/cyg_pub/devel/hdf5/hdf5-1.8.17-1.x86_64/src/hdf5-1.8.17=/usr/src/debug/hdf5-1.8.17-1
                      H5_CFLAGS:
                      AM_CFLAGS:
                       CPPFLAGS:
                    H5_CPPFLAGS:   -DNDEBUG -UH5_DEBUG_API
                    AM_CPPFLAGS:
               Shared C Library: yes
               Static C Library: no
  Statically Linked Executables: no
                        LDFLAGS:
                     H5_LDFLAGS:  -no-undefined
                     AM_LDFLAGS:
                Extra libraries: -lz -ldl -lm
                       Archiver: ar
                         Ranlib: ranlib
              Debugged Packages:
                    API Tracing: no

Languages:
----------
                        Fortran: no

                            C++: yes
                   C++ Compiler: /usr/bin/g++
                      C++ Flags: -ggdb -O2 -pipe -fdebug-prefix-map=/cygdrive/e/cyg_pub/devel/hdf5/hdf5-1.8.17-1.x86_64/build=/usr/src/debug/hdf5-1.8.17-1 -fdebug-prefix-map=/cygdrive/e/cyg_pub/devel/hdf5/hdf5-1.8.17-1.x86_64/src/hdf5-1.8.17=/usr/src/debug/hdf5-1.8.17-1
                   H5 C++ Flags:
                   AM C++ Flags:
             Shared C++ Library: yes
             Static C++ Library: no

Features:
---------
                  Parallel HDF5: no
             High Level library: yes
                   Threadsafety: no
            Default API Mapping: v18
 With Deprecated Public Symbols: yes
         I/O filters (external): deflate(zlib)
                            MPE: no
                     Direct VFD: no
                        dmalloc: no
Clear file buffers before write: yes
           Using memory checker: no
         Function Stack Tracing: no
      Strict File Format Checks: no
   Optimization Instrumentation: no
Bye...
Aborted (core dumped)

xmikos, 2017-03-23:
You should be able to install QSpectrumAnalyzer and soapy_power with `pip` on Windows and it should work (I have tried it on Windows 7 Pro X64). SciPy (which could be problematic) is now only optional dependency. Something like this should definitely work (only on 64-bit Windows, because there are no public builds of SoapySDR for 32-bit Windows):

1. install Python 3.6 (64-bit) from python.org ([download](https://www.python.org/ftp/python/3.6.1/python-3.6.1-amd64.exe)) and add Python to PATH
2. install SoapySDR libraries and drivers (bundled as part of Pothos SDR installer: [download](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D))
2. `pip install PyQt5`
3. `pip install QSpectrumAnalyzer`
4. `qspectrumanalyzer`

xmikos, 2017-04-03:
Can you try new installer (available on GitHub releases page) and follow procedure in README?

xmikos, 2017-04-15:
I am closing this issue since there has been no reaction. Latest version should work even on Windows 10, if you still has problems, send comment here and I will reopen it.

closed 2017-04-15 by xmikos