xmikos/qspectrumanalyzer #18: HackRF Sample Rate

derekchal, 2017-02-27:
I am unable to change the sample rate when using a hackrf.
Does this come into the same category as the ppm and crop issues?

( On my Intel i7 notebook the 20million sample rate is ok, but on an older desktop machine that sample rate cannot be supported)

Other than that, a great tool for a fast scan of a large frequency space.

xmikos, 2017-02-27:
Yes, definitely.

If you don't need extremely wide fast spectrum scan, you will be able to use my new universal [soapy_power](https://github.com/xmikos/soapy_power) backend soon. But it is not ready yet. I will release new stable version of QSpectrumAnalyzer when it is finished.

xmikos, 2017-03-14:
New version with soapy_power backend has been released, I am closing it for now.

closed 2017-03-14 by xmikos