xmikos/qspectrumanalyzer #21: File menu does not work on OS X

mehdideveloper, 2017-03-14:
Hi
In my installation, there is a "File" menu, but clicking it does not show or do anything (it does not have any sub-menus)
It does not show any errors on the terminal either.

Any ideas?

OS X 10.12.2
Python 3.6.0
PyQt4 (4.12) and pyqtgraph (0.10) and other prerequisites are installed.
Qt version: 4.8.7 (installed by brew)

xmikos, 2017-03-14:
There is very important *Settings* section in *File* menu. I don't have Mac OS X, so I can't debug it. But latest Git master already supports PyQt5, so can you try to uninstall PyQt4, install PyQt5 and try QSpectrumAnalyzer from Git? Maybe PyQt5 works better on Mac OS X...

mehdideveloper, 2017-03-14:
Well now I have another problem: the File menu is completely gone!
Here's what I did:
I removed PyQt4, installed PyQt5, pulled the latest code from github and set it up with
`sudo python3 setup.py install`
(I had the Qt5 already on my system)

xmikos, 2017-03-14:
Ok, I have found that this is known Mac OS X problem: http://stackoverflow.com/questions/2259886/pyqt-menubar-mac-osx-snow-leopard

I will try some workaround and publish it in new macosx git branch. Please let me then know if it has helped.

xmikos, 2017-03-14:
According to Qt documentation, *Settings* and *Quit* menu actions are automagically moved to Mac OS X specific *Application* menu. Settings will be renamed to *Preferences* there. No File menu will be displayed, because there are no other actions.

Are you sure that *Preferences* and *Quit* are not in Application menu?

Btw. can you please post me some screenshot?

mehdideveloper, 2017-03-14:
@xmikos  Thanks. You're right. 
It's located at "Python -> Preferences" 
Here is a screenshot:

[image]

closed 2017-03-21 by xmikos