xmikos/qspectrumanalyzer #23: make soapy_power optional

ZeroChaos-, 2017-03-23:
Would it be possible to allow function without soapy_power?  For example I want to use the hackrf_sweep backend.

xmikos, 2017-03-23:
Yes, it would be possible, but right now I am importing some classes from *soapy_power*.

I am developing QSpectrumAnalyzer with soapy_power as default universal backend so I didn't look at it as a problem. soapy_power is useful even on HackRF (hackrf_sweep is very fast, but also very coarse - 40 kHz bin size as a minimum, could not set longer integration time, etc.)

Is dependency on soapy_power (and therefore SoapySDR) really that problematic?

ZeroChaos-, 2017-03-23:
It's a bit problematic because it pulls in so many deps which are not otherwise needed.
For an example I've already spent about 3 hours trying to package up the deps that gentoo was missing for qspectrumanalyzer and I still have 2 to go.

Is it a crisis, no, I'm going to include support for it either way; but it's a huge amount of software to set up that the user might not even care about or need.

xmikos, 2017-03-23:
SoapySDR does have minimal dependencies, it is really lightweight if you don't install device drivers (which are optional and loaded at runtime). Much more lightweight than e.g. gnuradio-osmosdr (the only other similar universal SDR drivers framework).

xmikos, 2017-03-23:
But OK, I will make it optional (right now there is only one class from it used, so it is not a problem).

xmikos, 2017-03-23:
Done.

`soapy_power` dependency is still required in `setup.py` (I don't plan to remove it, as it is the default backend), but nevertheless QSpectrumAnalyzer will now work even without it being installed.

soapy_power backend will just fail silently when pressing Start button if there is no soapy_power module installed.

closed 2017-03-23 by xmikos

ZeroChaos-, 2017-03-24:
SoapySDR has deps, then soapy_power which needs simplesoapy and simplespectral, pyFFT being warmly recommended, etc.

Please don't misread what I'm saying.  You are doing great work, it is vastly appreciated, and I have to add all this to Pentoo (my livecd) and Gentoo eventually because people want it.  *But*, while adding it, it appeared that soapy_power could be an optional dep, and honestly my negative experiences with soapysdr lead me to ask that if it can be optional it be made optional.

ZeroChaos-, 2017-03-24:
awesome, thank you very much!

New tag please? :-)

xmikos, 2017-03-24:
I want to fix some known issues (and ideally also finish Windows installer) before next release. Maybe tomorrow.

Btw. may I humbly ask you to also package soapy_power (even if it is now optional)? Without it QSpectrumAnalyzer will be only usable on HackRF and RTL-SDR, but there are also many people with Airspy and LimeSDR (and also SDRplay, but I have problems with it for now, so I can't guarantee that it will work on SDRplay correctly yet).

ZeroChaos-, 2017-03-24:
I spent the last few hours packaging either things you wrote, or soapysdr related stuff.  The main thing I haven't yet packaged is all the soapy backends.... but given how funny it is to say soapy backends I'll probably do that tomorrow.

BTW, soapy_power makes me have a lot more faith in soapysdr than rx_power did, nicely done.  And thanks again, we are all working here to get awesome tools in the hands of the people that need them, your work is appreciated.