xmikos/qspectrumanalyzer #24: AttributeError: 'module' object has no attribute 'get_terminal_size'

super-schaf, 2017-03-23:
After hitting "Start" button, I managed to dwindle out most errors, I can't seem to get this one away;

Any ideas?

Thanks and greetings from a hill in Switzerland.

> Traceback (most recent call last):
>   File "/usr/local/bin/soapy_power", line 11, in <module>
>     load_entry_point('soapy-power==1.5.0', 'console_scripts', 'soapy_power')()
>   File "/usr/local/lib/python2.7/dist-packages/soapypower/__main__.py", line 258, in main
>     parser = setup_argument_parser()
>   File "/usr/local/lib/python2.7/dist-packages/soapypower/__main__.py", line 114, in setup_argument_parser
>     os.environ['COLUMNS'] = str(shutil.get_terminal_size().columns)
> AttributeError: 'module' object has no attribute 'get_terminal_size'
> 

Ubuntu 16.04.1

xmikos, 2017-03-23:
You are using old version of Python (Python 2.7). QSpectrumAnalyzer needs at least Python 3.3 (maybe even 3.4, I am not sure now if I didn't used some feature not found in 3.3 in latest changes).

xmikos, 2017-03-23:
How did you installed it? Did you followed README and used:

`pip3 install --user qspectrumanalyzer`

Or some other method? `pip3` is crucial here. If you by mistake used only `pip` instead, it will use old Python 2.7 on Ubuntu.

super-schaf, 2017-03-26:
Ah, well thanks for the prompt answer! 
Indeed I must have missed out the "3" from pip3... how very incompetent of myself. Sorry!

It works!! but one small question remains, though. 
from 700Mhz - 2700mhz bit a bin size of 40'000khz it took me just under 30 seconds to sweep once...
I have the latest hackrf firmware, hackrf_sweep works. Got pyfftw installed and all needed drivers.

I'm slightly clueless....again

maybe you have some insight into what may be causing it.

> [14:49:13]  Ignored exception:

    |==============================>>
    |  Traceback (most recent call last):
    |    File "/home/richard/.local/bin/qspectrumanalyzer", line 9, in <module>
    |      load_entry_point('QSpectrumAnalyzer==2.1.0', 'gui_scripts', 'qspectrumanalyzer')()
    |    File "/home/richard/.local/lib/python3.5/site-packages/qspectrumanalyzer/__main__.py", line 620, in main
    |      sys.exit(app.exec_())
    |    File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/widgets/GraphicsView.py", line 154, in paintEvent
    |      return QtGui.QGraphicsView.paintEvent(self, ev)
    |    File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/debug.py", line 93, in w
    |      printExc('Ignored exception:')
    |    --- exception caught here ---
    |    File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/debug.py", line 91, in w
    |      func(*args, **kwds)
    |    File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py", line 430, in paint
    |      path = self.getPath()
    |    File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py", line 412, in getPath
    |      self.path = self.generatePath(*self.getData())
    |    File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py", line 401, in generatePath
    |      path = fn.arrayToQPath(x, y, connect=self.opts['connect'])
    |    File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/functions.py", line 1378, in arrayToQPath
    |      arr[1:-1]['y'] = y
    |  ValueError: could not broadcast input array from shape (50692) into shape (50876)
    |==============================
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
<<

xmikos, 2017-03-26:
This error is with hackrf_sweep backend or soapy_power backend? It is known problem with hackrf_sweep backend (hackrf_sweep utility sometimes outputs incorrect number of bins, especially in heavy system load). There is already fix in QSpectrumAnalyzer Git master branch, there is just not new release yet.

super-schaf, 2017-03-26:
I may note that the errors only occurs when I try to sweep a large spectrum but it is rather slow even just scanning 100mhz (takes about 5 seconds) .Anyway, I don't want to bother you though with my little problems :) thanks for your support!

>   File "/home/richard/.local/lib/python3.5/site-packages/qspectrumanalyzer/plot.py", line 130, in update_plot
    self.curve.setData(data_storage.x, data_storage.y)
  File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 461, in setData
    self.updateItems()
  File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 487, in updateItems
    self.curve.setData(x=x, y=y, **curveArgs)
  File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py", line 307, in setData
    self.updateData(*args, **kargs)
  File "/home/richard/.local/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py", line 353, in updateData
    raise Exception("X and Y arrays must be the same shape--got %s and %s." % (self.xData.shape, self.yData.shape))
Exception: X and Y arrays must be the same shape--got (54110,) and (53884,).
[22:04:30]  Ignored exception:

xmikos, 2017-03-26:
This is not little problem ;-) But you still didn't tell me if this is with soapy_power or with hackrf_sweep? Judging by slower sweep time, it is soapy_power backend? Could you please send me whole log printed to terminal window?

xmikos, 2017-04-15:
I am closing this issue since there has been no reaction. If you still has this problem even with latest version of QSpectrumAnalyzer, please send me whole log printed to terminal window and I will reopen this issue.

closed 2017-04-15 by xmikos