xmikos/qspectrumanalyzer #25: hackrf_sweep

ZeroChaos-, 2017-03-24:
Got this when hitting stop
```
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/qspectrumanalyzer/backends/hackrf_sweep.py", line 139, in run
    buf = self.process.stdout.read(4)
AttributeError: 'NoneType' object has no attribute 'stdout'
zsh: abort      qspectrumanalyzer
```

xmikos, 2017-03-24:
This seems like that `hackrf_sweep` process never started (or more likely hackrf_sweep backend tried to read from it after stop). Strange, I can't reproduce it here, is it only something occasional or does it happen often?

ZeroChaos-, 2017-03-24:
great question, because now I can't get it to stop at all. the window is redrawing but completely non-responsive

xmikos, 2017-03-24:
Any news? Does it work or not?

I see you are using old Python 3.4. QSpectrumAnalyzer / soapy_power should work with Python 3.4, but I am running Python 3.6 so can't be completely sure. I will compile Python 3.4 and test it again, just to be sure.

michaellass, 2017-03-24:
I included a fix for this in my latest PR (#28).

xmikos, 2017-03-24:
@ZeroChaos- I have merged @michaellass' pull request, can you try it?

ZeroChaos-, 2017-04-04:
Using the 2.2.0 qspectrumanalyzer it's still extremely unresponsive, bordering on completely unresponsive.  it seems that qspectrumanalyzer takes one cpu to 100% and then pretty much locks up, even drawing starts to fail.  I can't reproduce the original issue, but I can't really say it's working in any useful sense of the word.  The terminal is scrolling as expected, whining about getting 9018 bins when expecting 10020, but always those exact numbers.

michaellass, 2017-04-04:
It's very easy to setup QSpectrumAnalyzer with the hackrf_sweep backup in a way where it is not able to cope with the amount of incoming data. Please try to either
- increase the frequency range (Stop-Start)
- increase the bin size
- set a sweep interval (e.g. 0.3 or 0.5 seconds)

The last option was added in my pull request in order to allow using QSpectrumAnalyzer with small frequency ranges and small bin sizes.

ZeroChaos-, 2017-04-04:
I cannot find the sweep interval option anywhere, so I can't set it :-)

You may be right about the problem of it just being overwhelmed.  I've tested with a lot of different settings and it seems that qspectrumanalyzer gets maxed out pretty easily with this.  Is there way the data parsing or drawing (I don't know what the slow part is) could be sped up?

michaellass, 2017-04-04:
> I cannot find the sweep interval option anywhere, so I can't set it :-)

[image]

Being precise it's not the real sweep interval but more of an arbitrary slowdown when using the hackrf_sweep backend (hackrf_sweep itself will still continue to run with the minimum interval possible but QSpectrumAnalyzer will drop sweeps).

closed 2017-04-15 by xmikos