xmikos/qspectrumanalyzer #26: ValueError

ZeroChaos-, 2017-03-24:
Starts to draw with hackrf_sweep then explodes, not really sure why, it was working last night on my system.  literally no idea what I changed.

```
zero@gato ~ % qspectrumanalyzer 
hackrf_sweep params:
{'bin_size': 40.0,
 'crop': 0,
 'device': 0,
 'gain': 0.0,
 'hops': 0,
 'interval': 0,
 'lna_gain': 0.0,
 'ppm': 0,
 'sample_rate': 20000000.0,
 'single_shot': False,
 'start_freq': 0.0,
 'stop_freq': 7240.0,
 'vga_gain': 0.0}

call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 7240 MHz
Stop with Ctrl-C
13.1 MiB / 1.000 sec = 13.1 MiB/second
11.8 MiB / 1.000 sec = 11.8 MiB/second
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/qspectrumanalyzer/data.py", line 54, in run
    result = self.task(*self.args, **self.kwargs)
  File "/usr/lib64/python3.4/site-packages/qspectrumanalyzer/data.py", line 133, in update_history
    self.history.append(data["y"])
  File "/usr/lib64/python3.4/site-packages/qspectrumanalyzer/data.py", line 24, in append
    self.buffer[-1] = data
ValueError: cannot copy sequence with size 180750 to array axis with dimension 181000
zsh: abort      qspectrumanalyzer
```

xmikos, 2017-03-24:
This is known problem with `hackrf_sweep` backend, it sometimes (in heavy load, etc.) outputs incorrect number of bins (some bins are simply lost). There is already [pull request](https://github.com/xmikos/qspectrumanalyzer/pull/22) with solution, I just didn't merged it yet.

xmikos, 2017-03-24:
It's merged, you can try master branch.

rct, 2017-03-24:
It would be beneficial to open an issue (or issues) in the hackrf repo so that all consumers of hackrf_sweep don't have to contend with the same problems.

xmikos, 2017-03-24:
@rtc problem is that I can't reproduce it on my system, *hackrf_sweep* is stable here. But more people has similar issue.

@michaellass can you please open issue in hackrf repo?

michaellass, 2017-03-24:
@xmikos Sure, I will do that and link it here.

ZeroChaos-, 2017-03-29:
please push a tag with this fix and feel free to close this ticket when you do.

xmikos, 2017-04-01:
Done.

closed 2017-04-01 by xmikos