xmikos/qspectrumanalyzer #27: soapy_power failure

ZeroChaos-, 2017-03-24:
Not sure exactly which codebase is blowing up but I get this while running qspectrumanalyzer with the soapy_power backend.  Again, it was working last night and this morning, no idea what I've done.

```
soapy_power params:
{'bandwidth': 20000000.0,
 'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 37,
 'hops': 0,
 'interval': 10.0,
 'ppm': 0,
 'sample_rate': 20000000.0,
 'single_shot': False,
 'start_freq': 2400.0,
 'stop_freq': 2500.0}

WARNING: Incorrect RX channel number, using channel 0 instead!
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.4/soapy_power", line 11, in <module>
    load_entry_point('soapy-power==1.5.0', 'console_scripts', 'soapy_power')()
  File "/usr/lib64/python3.4/site-packages/soapypower/__main__.py", line 301, in main
    output_format=args.format
  File "/usr/lib64/python3.4/site-packages/soapypower/power.py", line 38, in __init__
    force_sample_rate=force_sample_rate, force_bandwidth=force_bandwidth
  File "/usr/lib64/python3.4/site-packages/simplesoapy.py", line 73, in __init__
    self.sample_rate = sample_rate
  File "/usr/lib64/python3.4/site-packages/simplesoapy.py", line 167, in sample_rate
    real_sample_rate = closest(self.list_sample_rates(), sample_rate)
  File "/usr/lib64/python3.4/site-packages/simplesoapy.py", line 14, in closest
    return min(num_list, key=lambda x: abs(x - num))
ValueError: min() arg is an empty sequence
```

xmikos, 2017-03-24:
I don't know what is wrong with your installation :-( I have now even tried to compile Python 3.4 and use completely independent clean Python 3.4 virtualenv (just to be sure that it works) and I have found one incompatibility in latest version of soapy_power (previous versions were good, I will commit fix soon), but still I can't reproduce any of your issues :-/

Don't you have e.g. some wrong params in additional parameters in *Settings*? Can you try *soapy_power* from terminal (not from QSpectrumAnalyzer GUI)?

ZeroChaos-, 2017-04-04:
qspectrumanalyzer 2.2.0
soapy_power 1.6.0
simplesoapy 1.4.0
hackrf

I can test git of anything you like, but this is just me trying to run the defaults more or less.
'''
soapy_power params:
{'bandwidth': 20000000.0,
 'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 37.0,
 'hops': 0,
 'interval': 1.0,
 'ppm': 0,
 'sample_rate': 20000000.0,
 'single_shot': False,
 'start_freq': 2400.0,
 'stop_freq': 2500.0}

Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.4/soapy_power", line 11, in <module>
    load_entry_point('soapy-power==1.6.0', 'console_scripts', 'soapy_power')()
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 2627, in load_entry_point
    return ep.load()
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 2287, in load
    return self.resolve()
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 2293, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib64/python3.4/site-packages/soapypower/__main__.py", line 5, in <module>
    import simplesoapy
  File "/usr/lib64/python3.4/site-packages/simplesoapy.py", line 42, in <module>
    class Ranges(collections.abc.Sequence):
AttributeError: 'module' object has no attribute 'abc'
'''

This is tested with exact defaults, I just select soapy_power in settings and hit start
'''
soapy_power params:
{'bandwidth': 0.0,
 'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 37.0,
 'hops': 0,
 'interval': 1.0,
 'ppm': 0,
 'sample_rate': 2560000.0,
 'single_shot': False,
 'start_freq': 87.0,
 'stop_freq': 108.0}

Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.4/soapy_power", line 11, in <module>
    load_entry_point('soapy-power==1.6.0', 'console_scripts', 'soapy_power')()
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 2627, in load_entry_point
    return ep.load()
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 2287, in load
    return self.resolve()
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 2293, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib64/python3.4/site-packages/soapypower/__main__.py", line 5, in <module>
    import simplesoapy
  File "/usr/lib64/python3.4/site-packages/simplesoapy.py", line 42, in <module>
    class Ranges(collections.abc.Sequence):
AttributeError: 'module' object has no attribute 'abc'
'''

xmikos, 2017-04-05:
You have too old version of `simplesoapy`. Compatibility with old Python 3.4 has been fixed in latest version 1.4.1.

ZeroChaos-, 2017-04-06:
upgraded to 1.4.1

```
zero@gato simplesoapy % qspectrumanalyzer                                                                                                                               (git)-[master] 
soapy_power params:
{'bandwidth': 20000000.0,
 'bin_size': 3.0,
 'crop': 0.0,
 'device': '',
 'gain': 37.0,
 'hops': 0,
 'interval': 1.0,
 'ppm': 0,
 'sample_rate': 20000000.0,
 'single_shot': False,
 'start_freq': 100.0,
 'stop_freq': 470.0}

linux; GNU C++ version 4.9.4; Boost_106200; UHD_003.008.005-0-unknown

[INFO] Opening HackRF device instance {device=HackRF One, driver=hackrf, part_id=a000cb3c005f434b, serial=000000000000000015d463dc38437825, version=2017.02.1}...
INFO: Using device: HackRF One
WARNING: number of FFT bins should be even, changing to 6668
INFO: repeats: 158
INFO: samples: 1053544 (time: 0.05268 s)
INFO: max_buffer_size (samples): 13107200 (repeats: 1965.69, time: 0.65536 s)
INFO: buffer_size (samples): 1179648 (repeats: 176.91, time: 0.05898 s)
INFO: buffer_repeats: 1
INFO: overlap: 0.00000
INFO: bin_size: 2999.40 Hz
INFO: bins: 6668
INFO: bins (after crop): 6668
INFO: sample_rate: 20.000 MHz
INFO: sample_rate (after crop): 20.000 MHz
INFO: freq_range: 370.000 MHz
INFO: hopping: YES
INFO: hop_size: 20.000 MHz
INFO: hops: 19
INFO: min_center_freq: 110.000 MHz
INFO: max_center_freq: 470.000 MHz
INFO: min_freq (after crop): 100.000 MHz
INFO: max_freq (after crop): 480.000 MHz
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
OOOERROR: len(x_axis) != len(y_axis)
OOOOOERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
OERROR: len(x_axis) != len(y_axis)
OOOERROR: len(x_axis) != len(y_axis)
OERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
OERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
OOERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
OOOOERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
ERROR: len(x_axis) != len(y_axis)
OTraceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/qspectrumanalyzer/plot.py", line 130, in update_plot
    self.curve.setData(data_storage.x, data_storage.y)
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 461, in setData
    self.updateItems()
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 487, in updateItems
    self.curve.setData(x=x, y=y, **curveArgs)
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py", line 307, in setData
    self.updateData(*args, **kargs)
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/PlotCurveItem.py", line 353, in updateData
    raise Exception("X and Y arrays must be the same shape--got %s and %s." % (self.xData.shape, self.yData.shape))
Exception: X and Y arrays must be the same shape--got (126711,) and (126692,).
zsh: abort      qspectrumanalyzer
ERROR: Error writing to output file: [Errno 32] Broken pipe                                                                                                                             
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/soapypower/writer.py", line 122, in write
    pwr_array
  File "/usr/lib64/python3.4/site-packages/soapypower/writer.py", line 90, in write
    f.write(pwr_array.tobytes())
BrokenPipeError: [Errno 32] Broken pipe
```

then repeating this until I unplug my hackrf
```
ERROR: Error writing to output file: [Errno 32] Broken pipe
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/soapypower/writer.py", line 122, in write
    pwr_array
  File "/usr/lib64/python3.4/site-packages/soapypower/writer.py", line 90, in write
    f.write(pwr_array.tobytes())
```

xmikos, 2017-04-27:
Can you please test latest QSpectrumAnalyzer Git master branch? Latest commit can hopefully fix this error.

Please upgrade also SimpleSoapy and soapy_power befor testing it.

ZeroChaos-, 2017-07-27:
both 2.2.0 and git leave me with this:

Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.4/soapy_power", line 11, in <module>
    load_entry_point('soapy-power==1.6.1', 'console_scripts', 'soapy_power')()
  File "/usr/lib64/python3.4/site-packages/soapypower/__main__.py", line 373, in main
    max_threads=args.max_threads, max_queue_size=args.max_queue_size
  File "/usr/lib64/python3.4/site-packages/soapypower/power.py", line 296, in sweep
    psd_future, acq_time_start, acq_time_stop = self.psd(freq)
  File "/usr/lib64/python3.4/site-packages/soapypower/power.py", line 256, in psd
    self.device.read_stream_into_buffer(self._buffer)
  File "/usr/lib64/python3.4/site-packages/simplesoapy.py", line 484, in read_stream_into_buffer
    res.ret, SoapySDR.errToStr(res.ret)
RuntimeError: Unhandled readStream() error: -1 (TIMEOUT)

guo1206, 2017-09-05:
@ZeroChaos- 
both 2.2.0 and git leave me with this:

Traceback (most recent call last):
File "/usr/lib/python-exec/python3.4/soapy_power", line 11, in 
load_entry_point('soapy-power==1.6.1', 'console_scripts', 'soapy_power')()
File "/usr/lib64/python3.4/site-packages/soapypower/main.py", line 373, in main
max_threads=args.max_threads, max_queue_size=args.max_queue_size
File "/usr/lib64/python3.4/site-packages/soapypower/power.py", line 296, in sweep
psd_future, acq_time_start, acq_time_stop = self.psd(freq)
File "/usr/lib64/python3.4/site-packages/soapypower/power.py", line 256, in psd
self.device.read_stream_into_buffer(self._buffer)
File "/usr/lib64/python3.4/site-packages/simplesoapy.py", line 484, in read_stream_into_buffer
res.ret, SoapySDR.errToStr(res.ret)
RuntimeError: Unhandled readStream() error: -1 (TIMEOUT)

Please whether this issue has been resolved ?If it has,how to solve the issue?

tdbone1, 2018-03-09:
windows 10 pro x86_64 on intel i3_4005u (spring creators update "pre") with E4000 plugged in usb 2.0
C:\Program Files\QSpectrumAnalyzer>soapy_power --info
WARNING: Incorrect RX channel number, using channel 0 instead!
Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966.py", line 2, in <module>
  File "soapypower\__main__.py", line 301, in main
  File "soapypower\__main__.py", line 106, in device_info
  File "site-packages\simplesoapy.py", line 384, in get_frequency_range
IndexError: tuple index out of range
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

C:\Program Files\QSpectrumAnalyzer>soapy_power --info
WARNING: Incorrect RX channel number, using channel 0 instead!
Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966.py", line 2, in <module>
  File "soapypower\__main__.py", line 301, in main
  File "soapypower\__main__.py", line 106, in device_info
  File "site-packages\simplesoapy.py", line 384, in get_frequency_range
IndexError: tuple index out of range
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

ZeroChaos-, 2018-08-21:
I have no explanation, but I am no longer able to reproduce this issue.

closed 2018-08-21 by ZeroChaos-