xmikos/qspectrumanalyzer #29: Gain sliders for hackrf

vsboost, 2017-04-03:
Hi there,

Are you able to include
Slider for IF LNA gain 0-40db in 8db steps
Slider for Baseband VGA gain 0-62db in 2db steps
Button for RF amp 0db = OFF or 14db = ON

Cheers

xmikos, 2017-04-03:
Hello,

you can set gain to -1 in GUI (displayed as "auto") and then add `--specific-gains` option to additional backend parameters in *Settings*. E.g.: `--specific-gains LNA=28,VGA=12,AMP=0`.

This is for `soapy_power` backend. For `hackrf_sweep` backend you can use options `-l` (LNA) and `-g` (VGA), but I am afraid this would not work (hackrf_sweep backend always sets gain be itself, even if you set it to -1.

xmikos, 2017-04-03:
I will change hackrf_sweep backend to also work like soapy_power backend and let you know here.

xmikos, 2017-04-03:
It's implemented in latest two commits (git master branch), please try it out and let me know how it works for you.

vsboost, 2017-04-04:
Thanks for this.

closed 2017-04-06 by xmikos