xmikos/qspectrumanalyzer #30: soapy-power supposed to be optional in 2.2.0 but is not

ZeroChaos-, 2017-04-04:
Not only is soapy-power not optional, it seems to be checking soapy-power's deps from qspectrumanalyzer
'''
zero@gato qspectrumanalyzer % qspectrumanalyzer                                                                                                                                                    (git)-[master] 
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 655, in _build_master
    ws.require(__requires__)
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 854, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (SimpleSoapy 1.3.0 (/usr/lib64/python3.4/site-packages), Requirement.parse('simplesoapy>=1.4.0'), {'soapy-power'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.4/qspectrumanalyzer", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 3015, in <module>
    @_call_aside
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 2999, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 3028, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 657, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 670, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib64/python3.4/site-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'simplesoapy>=1.4.0' distribution was not found and is required by soapy-power
'''

xmikos, 2017-04-06:
It is optional, QSpectrumAnalyzer will actually work if you start it without `soapy_power` being installed. But soapy_power is still required in `setup.py` (and I really don't want to change that), which means startup scripts generated by setup.py will check if it is installed. You can avoid it by starting QSpectrumAnalyzer like this:

    python3 -m qspectrumanalyzer

Or write your own startup script:

    #!/usr/bin/env python3
    from qspectrumanalyzer.__main__ import main
    main()

xmikos, 2017-04-15:
I am closing this issue. I hope that what I have recommended in my previous comment worked for you. If it did not, then please let me know and I will reopen this issue.

closed 2017-04-15 by xmikos