xmikos/qspectrumanalyzer #31: No devices Found -- soapy_power

GinesGarcia, 2017-04-25:
Hi,
I'm trying to use the qspectrumanalyzer with my USRP card but when I run it, no device is found by soapy_power. I run directly soapy_power command to detect my device but i got the same answer:

    soapy_power: error: No devices found!

Software components such as srsUE or OAI detect correctly the USRP card.

Any idea about how to solve this problem?

Thanks in advance.

xmikos, 2017-04-25:
Did you installed SoapySDR drivers for USRP (SoapyUHD)?

GinesGarcia, 2017-04-25:
Thanks!
I forgot to download the image :)

xmikos, 2017-04-25:
So does it work with USRP? It should work, but I didn't have chance to test it (I don't have USRP).

GinesGarcia, 2017-04-25:
It seems like soapy_power detects the USRP ("soapy_power --info" command detects correctly my USRP device ) but when I start scanning with qspectrumanalyzer I get the following error:

  soapy_power params:
{'bandwidth': 0.0,
 'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 0,
 'hops': 0,
 'interval': 10.0,
 'ppm': 0,
 'sample_rate': 2560000.0,
 'single_shot': False,
 'start_freq': 42.0,
 'stop_freq': 60.0}

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.001.001-release

[INFO] Detected Device: B210
[INFO] Operating over USB 3.
[INFO] Initialize CODEC control...
[INFO] Initialize Radio control...
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Setting master clock rate selection to 'automatic'.
[INFO] Asking for clock rate 16.000000 MHz... 
[INFO] 
Actually got clock rate 16.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
WARNING: Sample rate 2560000.0 Hz is not supported, setting it to 2666666.6666666665 Hz!
[INFO] Asking for clock rate 42.666672 MHz... 
[INFO] 
Actually got clock rate 42.666672 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
INFO: Using device: B210
WARNING: number of FFT bins should be even, changing to 268
[INFO] Asking for clock rate 42.666672 MHz... 
[INFO] OK
[INFO] Asking for clock rate 42.666672 MHz... 
[INFO] OK
INFO: repeats: 14215
INFO: samples: 3809620 (time: 1.42861 s)
INFO: max_buffer_size (samples): 13108172 (repeats: 48911.09, time: 4.91556 s)
INFO: buffer_size (samples): 3810016 (repeats: 14216.48, time: 1.42876 s)
INFO: buffer_repeats: 1
INFO: overlap: 0.00000
INFO: bin_size: 9950.25 Hz
INFO: bins: 268
INFO: bins (after crop): 268
INFO: sample_rate: 2.667 MHz
INFO: sample_rate (after crop): 2.667 MHz
INFO: freq_range: 18.000 MHz
INFO: hopping: YES
INFO: hop_size: 2.667 MHz
INFO: hops: 7
INFO: min_center_freq: 43.333 MHz
INFO: max_center_freq: 59.333 MHz
INFO: min_freq (after crop): 42.000 MHz
INFO: max_freq (after crop): 60.667 MHz
INFO: Total time: 0.000 s
Traceback (most recent call last):
  File "/usr/local/bin/soapy_power", line 9, in <module>
    load_entry_point('soapy-power==1.6.0', 'console_scripts', 'soapy_power')()
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.0-py3.4.egg/soapypower/__main__.py", line 367, in main
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.0-py3.4.egg/soapypower/power.py", line 296, in sweep
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.0-py3.4.egg/soapypower/power.py", line 224, in psd
  File "/usr/local/lib/python3.4/dist-packages/SimpleSoapy-1.4.1-py3.4.egg/simplesoapy.py", line 160, in freq
    return self.device.getFrequency(SoapySDR.SOAPY_SDR_RX, self._channel, 'RF')
  File "/usr/lib/python3/dist-packages/SoapySDR.py", line 695, in getFrequency
    def getFrequency(self, *args): return _SoapySDR.Device_getFrequency(self, *args)
RuntimeError: map::at

xmikos, 2017-04-25:
It looks like USRP in SoapySDR is not providing default `RF` tunable element... could you please post output of `soapy_power --info`?

GinesGarcia, 2017-04-25:
soapy_power --info
```
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.001.001-release

[WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
  setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
[INFO] Detected Device: B210
[INFO] Operating over USB 3.
[INFO] Initialize CODEC control...
[INFO] Initialize Radio control...
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Setting master clock rate selection to 'automatic'.
[INFO] Asking for clock rate 16.000000 MHz... 
[INFO] 
Actually got clock rate 16.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
Selected device: B210
  Available RX channels:
    0, 1
  Available antennas:
    TX/RX, RX2
  Available tunable elements:
    RF, BB
  Available amplification elements:
    PGA
  Available device settings:
  Available stream arguments:
    peak ... Peak value - The peak value for scaling in complex byte mode. (default: 1.0)
    spp ... Samples per packet - The number of samples per packet. (default: 0)
    recv_frame_size ... Receive frame buffer size - The size an individual datagram or frame in bytes. (default: )
    num_recv_frames ... Receive number of buffers - The number of available buffers. (default: )
    WIRE ... Bus format - The format of samples over the bus. (default: )
  Allowed gain range [dB]:
    0.00 - 76.00
  Allowed frequency range [MHz]:
    42.00 - 6008.00
  Allowed sample rates [MHz]:
    0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.03, 0.04, 0.04, 0.04, 0.04, 0.04,
    0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04,
    0.04, 0.04, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.05, 0.06,
    0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.07, 0.07, 0.07, 0.07, 0.07, 0.07, 0.07,
    0.07, 0.07, 0.07, 0.07, 0.07, 0.07, 0.07, 0.07, 0.07, 0.07, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08,
    0.08, 0.08, 0.08, 0.09, 0.09, 0.09, 0.09, 0.09, 0.09, 0.09, 0.09, 0.09, 0.09, 0.10, 0.10, 0.10, 0.10, 0.10, 0.10,
    0.10, 0.10, 0.11, 0.11, 0.11, 0.11, 0.11, 0.11, 0.11, 0.12, 0.12, 0.12, 0.12, 0.12, 0.12, 0.13, 0.13, 0.13, 0.13,
    0.13, 0.13, 0.13, 0.13, 0.13, 0.14, 0.14, 0.14, 0.14, 0.14, 0.14, 0.14, 0.14, 0.15, 0.15, 0.15, 0.15, 0.15, 0.15,
    0.15, 0.16, 0.16, 0.16, 0.16, 0.16, 0.16, 0.16, 0.17, 0.17, 0.17, 0.17, 0.17, 0.18, 0.18, 0.18, 0.18, 0.18, 0.19,
    0.19, 0.19, 0.19, 0.20, 0.20, 0.20, 0.20, 0.21, 0.21, 0.21, 0.21, 0.22, 0.22, 0.22, 0.23, 0.23, 0.23, 0.24, 0.24,
    0.24, 0.25, 0.25, 0.25, 0.26, 0.26, 0.27, 0.27, 0.28, 0.28, 0.29, 0.29, 0.30, 0.30, 0.31, 0.31, 0.32, 0.33, 0.33,
    0.34, 0.35, 0.36, 0.36, 0.37, 0.38, 0.39, 0.40, 0.41, 0.42, 0.43, 0.44, 0.46, 0.47, 0.48, 0.50, 0.52, 0.53, 0.55,
    0.57, 0.59, 0.62, 0.64, 0.67, 0.70, 0.73, 0.76, 0.80, 0.84, 0.89, 0.94, 1.00, 1.07, 1.14, 1.23, 1.33, 1.45, 1.60,
    1.78, 2.00, 2.29, 2.67, 3.20, 4.00, 5.33, 8.00, 16.00
  Allowed bandwidths [MHz]:
    0.20, 56.00
```

soapy_power --detect
```
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.001.001-release

[WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
  setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
Traceback (most recent call last):
  File "/usr/local/bin/soapy_power", line 9, in <module>
    load_entry_point('soapy-power==1.6.0', 'console_scripts', 'soapy_power')()
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.0-py3.4.egg/soapypower/__main__.py", line 295, in main
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.0-py3.4.egg/soapypower/__main__.py", line 72, in detect_devices
  File "/usr/local/lib/python3.4/dist-packages/SimpleSoapy-1.4.1-py3.4.egg/simplesoapy.py", line 37, in detect_devices
    d_str.append('label={}'.format(d['label']))
KeyError: 'label'
```

xmikos, 2017-04-25:
Hmm, `RF` is provided in list of tunable elements, so I don't know where the problem could be :-(

Second issue seems to be that SoapyUHD driver is not providing `label`. Looking at [SoapyUHD GitHub repo](https://github.com/pothosware/SoapyUHD), this has been fixed in Git master branch. In the mean time I have made `label` optional in [SimpleSoapy](https://github.com/xmikos/simplesoapy), you can install it from Git master branch with `pip` and try it. If you need more help how to do it, just tell me.

GinesGarcia, 2017-04-25:
By using the master branch of SopyUHD the second error was fixed but I'm still not able to run qspectrumanalyzer (RuntimeError: map::at)

xmikos, 2017-04-26:
@GinesGarcia Can you please try to compile this branch of SoapyUHD? https://github.com/pothosware/SoapyUHD/tree/no_tune_result

It should fix it...

GinesGarcia, 2017-04-26:
By using this branch the problem persists. I have checked that SoapySDRUtil command detects correctly the USRP (also uhd_find_devices command) but when I start scanning with "qspectrumanalyzer", I got the same error.

Any suggestions?

Thanks in advance

xmikos, 2017-04-26:
I have forwarded your post to upstream issue pothosware/SoapyUHD#7. Just to be certain, are you sure that you have built *no_tune_result* branch (could you check that commit https://github.com/pothosware/SoapyUHD/commit/0542d3e6125e6f3bfad7bec2e622412d55c8ecf5 is included)?

GinesGarcia, 2017-04-26:
I think so, I have followed this steps:
1. git checkout no_tune_result
2. git pull
3. rm -rf build
4. mkdir build && cd build
5. cmake .. && make && sudo make install 
6. ~/.local/bin/qspectrumanalyzer

I have also checked that the commit that you mentioned before is included

xmikos, 2017-04-26:
Great, let's see what @guruofquality could tell us ;-)

guruofquality, 2017-04-27:
I tested this with a simple command `python -c "import SoapySDR; d = SoapySDR.Device(); d.getFrequency(SoapySDR.SOAPY_SDR_RX, 0, 'RF')"`. The branch I mentioned at least fixes the error for the python snippet above. So I merged that into master since its otherwise harmless. If the error persists, I'm going to need a traceback, because I don't where where the map::at() lookup is failing.

xmikos, 2017-04-27:
I have changed SimpleSoapy to use setFrequency / getFrequency without specifying RF tunable element (main reason has been to allow tunning of LimeSDR to < 30 MHz).

@GinesGarcia can you please try it with this new version of SimpleSoapy (and soapy_power)?

GinesGarcia, 2017-04-27:
Hi,
after pull the new version of soapy_power I'm getting the following error:

```
soapy_power --info
Traceback (most recent call last):
  File "/usr/local/bin/soapy_power", line 11, in <module>
    load_entry_point('soapy-power==1.6.1', 'console_scripts', 'soapy_power')()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "build/bdist.linux-x86_64/egg/soapypower/__main__.py", line 5, in <module>
  File "build/bdist.linux-x86_64/egg/simplesoapy.py", line 3, in <module>
ImportError: No module named abc

```

xmikos, 2017-04-27:
You are starting it with Python 2, but it needs at least Python 3.4. Didn't you installed it with Python 2 by mistake? In that case startup script created by setup.py will be wrongly calling Python 2.

xmikos, 2017-04-27:
(both simplesoapy, soapy_power and qspectrumanalyzer needs at least Python 3.4, it is in README)

GinesGarcia, 2017-04-27:
Sorry, I forgot to build/install soapy_power with python 3 :/
Now, when running qspectrumanalyzer I get this error:
```
~/.local/bin/qspectrumanalyzer

(process:13381): Gtk-WARNING **: Locale not supported by C library.
	Using the fallback 'C' locale.
soapy_power params:
{'bandwidth': 0.0,
 'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 0,
 'hops': 0,
 'interval': 10.0,
 'ppm': 0,
 'sample_rate': 2560000.0,
 'single_shot': False,
 'start_freq': 42.0,
 'stop_freq': 60.0}

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.001.001-release

[INFO] Detected Device: B210
[INFO] Operating over USB 3.
[INFO] Initialize CODEC control...
[INFO] Initialize Radio control...
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Setting master clock rate selection to 'automatic'.
[INFO] Asking for clock rate 16.000000 MHz... 
[INFO] 
Actually got clock rate 16.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
WARNING: Sample rate 2560000.0 Hz is not supported, setting it to 2666666.6666666665 Hz!
[INFO] Asking for clock rate 42.666672 MHz... 
[INFO] 
Actually got clock rate 42.666672 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
INFO: Using device: B210
WARNING: number of FFT bins should be even, changing to 268
[INFO] Asking for clock rate 42.666672 MHz... 
[INFO] OK
[INFO] Asking for clock rate 42.666672 MHz... 
[INFO] OK
INFO: repeats: 14215
INFO: samples: 3809620 (time: 1.42861 s)
INFO: max_buffer_size (samples): 13108172 (repeats: 48911.09, time: 4.91556 s)
INFO: buffer_size (samples): 3810016 (repeats: 14216.48, time: 1.42876 s)
INFO: buffer_repeats: 1
INFO: overlap: 0.00000
INFO: bin_size: 9950.25 Hz
INFO: bins: 268
INFO: bins (after crop): 268
INFO: sample_rate: 2.667 MHz
INFO: sample_rate (after crop): 2.667 MHz
INFO: freq_range: 18.000 MHz
INFO: hopping: YES
INFO: hop_size: 2.667 MHz
INFO: hops: 7
INFO: min_center_freq: 43.333 MHz
INFO: max_center_freq: 59.333 MHz
INFO: min_freq (after crop): 42.000 MHz
INFO: max_freq (after crop): 60.667 MHz
OWARNING: readStream returned only 453 samples, but buffer size is 2044!
ERROR: Error writing to output file: 'numpy.ndarray' object has no attribute 'tobytes'
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 122, in write
    pwr_array
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 90, in write
    f.write(pwr_array.tobytes())
AttributeError: 'numpy.ndarray' object has no attribute 'tobytes'
OWARNING: readStream returned only 453 samples, but buffer size is 2044!
ERROR: Error writing to output file: 'numpy.ndarray' object has no attribute 'tobytes'
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 122, in write
    pwr_array
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 90, in write
    f.write(pwr_array.tobytes())
AttributeError: 'numpy.ndarray' object has no attribute 'tobytes'
OWARNING: readStream returned only 453 samples, but buffer size is 2044!
ERROR: Error writing to output file: 'numpy.ndarray' object has no attribute 'tobytes'
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 122, in write
    pwr_array
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 90, in write
    f.write(pwr_array.tobytes())
AttributeError: 'numpy.ndarray' object has no attribute 'tobytes'
OWARNING: readStream returned only 453 samples, but buffer size is 2044!
OWARNING: readStream returned only 453 samples, but buffer size is 2044!
ERROR: Error writing to output file: 'numpy.ndarray' object has no attribute 'tobytes'
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 122, in write
    pwr_array
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 90, in write
    f.write(pwr_array.tobytes())
AttributeError: 'numpy.ndarray' object has no attribute 'tobytes'
^C
ggarcia@bichobola:~/soapy_power$ ERROR: Error writing to output file: 'numpy.ndarray' object has no attribute 'tobytes'
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 122, in write
    pwr_array
  File "/usr/local/lib/python3.4/dist-packages/soapy_power-1.6.1-py3.4.egg/soapypower/writer.py", line 90, in write
    f.write(pwr_array.tobytes())
AttributeError: 'numpy.ndarray' object has no attribute 'tobytes'
```

xmikos, 2017-04-27:
I see two problems here:

1. your numpy version doesn't provide `tobytes()` method. Maybe you have too old version of NumPy? Looking at NumPy documentation, you need at least version 1.9.0 (release in september 2014).
2. there is some problem reading samples (`WARNING: readStream returned only 453 samples, but buffer size is 2044!`), but first fix problem 1 before we look into this

xmikos, 2017-04-27:
I see that both Ubuntu Xenial (16.04 LTS) and Ubuntu Yaketty (16.10) have good version of NumPy (1.11), but Debian Jessie has too old version (1.8).

GinesGarcia, 2017-04-27:
Ok, now the application is working but I only get information in the time/frequency chart and I'm still getting some errors (log file attached).

[log.txt](https://github.com/xmikos/qspectrumanalyzer/files/960916/log.txt)

Thanks in advance,
Ginés.

xmikos, 2017-04-27:
@GinesGarcia could you please run `soapy_power -O output.csv --debug` and post me both `output.csv` file and console log?

GinesGarcia, 2017-04-27:
Here are the requested files (sorry for the output in txt but it is not possible to attach files in csv)

[soapy_log.txt](https://github.com/xmikos/qspectrumanalyzer/files/960951/soapy_log.txt)
[output.txt](https://github.com/xmikos/qspectrumanalyzer/files/960950/output.txt)

xmikos, 2017-04-27:
Well, this is strange, output looks good (there is still that `WARNING: readStream returned only 453 samples, but buffer size is 2044!`, but it only happens once, so there will be some missed samples, but overall it seems to work).

Please try similar setting like you have used in QSpectrumAnalyzer, so e.g.:

`soapy_power -O output.csv --debug -f 42M:60M -B 10k -r 2.56M -T 10 -u 10 --even`

And post me results (files will be much larger).

GinesGarcia, 2017-04-27:
[soapy_log2.txt](https://github.com/xmikos/qspectrumanalyzer/files/961127/soapy_log2.txt)
[output2.txt](https://github.com/xmikos/qspectrumanalyzer/files/961128/output2.txt)

xmikos, 2017-04-27:
@GinesGarcia I think I have found the issue, it seems to be problem with floating point precission leading to generating different number of bins for x axis (because `numpy.arange()` is used). To be fair somebody has already informed me about this potential issue (see https://github.com/xmikos/soapy_power/issues/2#issuecomment-289283142), but I didn't get time to look into that yet. I will do that today.

In the mean time can you please try QSpectrumAnalyzer with some different sample rate (now 2.66666 MHz has been used, because 2.560 MHz seems not to be supported by USRP)? Just to be sure that this is really the issue.

xmikos, 2017-04-27:
Can you please test your original settings (which gave you error) in latest QSpectrumAnalyzer Git master branch? Latest commit should hopefully fix it.