xmikos/qspectrumanalyzer #32: Windows Crash

gioreva, 2017-04-26:
Hi

I have Pyton 3.6 32bit (My CPU is intel i7)
NooElec RTL2832U & R820T RF key
PothosGUI vc14-x64  (Cubic SDR work well)
Zadig 2.3 (Bulk-in, Interface (Interface 0) -> WinUSB (v6.1.7600.16385)
Windows 7, 64bit

On QspectrumAnalyzer 2.2.0
rtl_power Device=0, wen start, it crash
rtl_power_fftw,  it crash
rx_power, it crash
soapy_power, not crash, but start stop immediatly and notyng appears.

Why is this so?

xmikos, 2017-04-26:
Also make sure that you have specified `driver=rtlsdr` in QSpectrumAnalyzer settings (if not specified, audio driver is used by default on Windows).

gioreva, 2017-04-26:
There are a document for how to configure ?
Where I add "driver=rtlsdr" ?
Additional parameter "--even --fft-window boxcar --remove-dc driver=rtlsdr" on "soapy_power" BackEnd

This "soapy_power --detect"
https://pastebin.com/EhTmNwXm
C:\Program Files\PothosSDR\lib\SoapySDR\modules0.5-2\bladeRFSupport.dll Exist on My PC

This "soapy_power -d driver=rtlsdr --info"
https://pastebin.com/HjehseyW

xmikos, 2017-04-26:
You should not add `driver=rtlsdr` to *Additional parameters* box, but to *Device* box. If you click on  **?** button besides *Device* box, it will show you all detected devices (same as `soapy_power --detect`) and you simply copy appropriate device string to *Device* box (but `driver=DEVICE_DRIVER_NAME` is enough).

But looking at your output of `soapy_power --detect`, there is definitely something wrong with your SoapySDR / Pothos installation. It could not load device-specific libraries (e.g. rtlsdrSupport.dll), so it will not work. Try to reinstall SoapySDR / Pothos bundle.

xmikos, 2017-04-26:
Btw. how did you installed QSpectrumAnalyzer / soapy_power? With installer or manually with `pip`?

gioreva, 2017-04-26:
installer

gioreva, 2017-04-26:
Removed and replaced pothos SDR
It Cannot edit PATH, "too long" error.
CubicSDR work well

C:\Program Files\QSpectrumAnalyzer>soapy_power --detect
Detected SoapySDR devices:
  No devices found!

Zadig Driver is Ok

gioreva, 2017-04-26:
Ok, adding PATH, now Work
But why Y is DB ?
I cannnot measure value amplitude with the key ? (dbm or sd/uV)
I've seen dBm screens, only with hackRF?

xmikos, 2017-04-26:
@gioreva it is still the same, dBm unit has been changed to dB because it could be misleading (without calibration, you can't be sure about the absolute value in dBm).

gioreva, 2017-04-26:
It worked a bit, now nothing appears. No crash.
https://pastebin.com/dBxYVXMB

I ordered hackRF, I need a spectrum analyzer for EMC measurements.
But I need a calibrated measurement in dB/uV
Also compensating antenna gain.
Also calibrated output, for calculate antenna gain.
Is there a chance that this program evolves in that sense?

xmikos, 2017-04-26:
It has been HackRF people who changed dBm label to dB label, see commit https://github.com/xmikos/qspectrumanalyzer/commit/70b921992764ca8988564001d49041bf2f5e1594:

> **Changed labels from dBm to dB.**
> hackrf_sweep outputs dBFS, but rtl_power outputs dB with some weird offset.
> In no case would dBm be correct without a calibration option.
> The relative "dB" label should be acceptable for all cases.

So if somebody do calibration, we can change it back to dBm. But every supported device would need separate calibration.

gioreva, 2017-04-27:
For calibrating input, I'm looking for a calibrated generator.
To calibrate the output, just connect input with output, you make a spam and analyze all frequency gain.
But already so is a strange thing.
I read -75db signal with gain 0
I put gain to 10, i expected, same -75db. -65 if you don't  consider it,  instead i read -87db

gioreva, 2017-04-27:
But you menage Output ?
I do not understand what the setup parameters are

xmikos, 2017-04-27:
I am sorry, I do not understand what you mean by "menage Output"?

I can describe you what exactly is soapy_power backend doing:

1. tune to center frequency for given frequency hop (given by frequency plan, if you run soapy_power with `--debug` option, you can see it in console log)
2. collect I/Q samples (number of samples is given by buffer size, again see soapy_power console log)
3. compute power spectral density (PSD) with Welch's method (see [scipy.signal.welch](https://scipy.github.io/devdocs/generated/scipy.signal.welch.html#scipy.signal.welch) documentation). Default FFT overlap is 50% and Hann window is used (but with `--fft-window boxcar`, rectangular window is used instead).
4. if you have specified `--remove-dc` option, central point is interpolated from closest points to remove DC bias (this is done after computing PSD, another alternative is to use `--detrend constant` option, which removes mean value from raw data to cancel DC bias)

xmikos, 2017-04-27:
In general, you can look at `soapy_power --help` to see what each option does. Or click to **?** button besides *Additional parameters* box in QSpectrumAnalyzer settings (it will also show you soapy_power help).

xmikos, 2017-04-27:
Also by default log scale is used, it is computed like this:
    pwr_array = 10 * numpy.log10(pwr_array)

xmikos, 2017-04-27:
> But already so is a strange thing.
> I read -75db signal with gain 0
> I put gain to 10, i expected, same -75db. -65 if you don't consider it, instead i read -87db

This has been error in SimpleSoapy, which didn't allow setting gain to 0 dB (some default value has been used instead). I have now fixed it in latest SimpleSoapy version 1.5.1.

gioreva, 2017-04-28:
From https://github.com/mossmann/hackrf/wiki/HackRF-One
Transmit Power

HackRF One's absolute maximum TX power varies by operating frequency:
10 MHz to 2150 MHz: 5 dBm to 15 dBm, generally increasing as frequency decreases
2150 MHz to 2750 MHz: 13 dBm to 15 dBm
2750 MHz to 4000 MHz: 0 dBm to 5 dBm, increasing as frequency decreases
4000 MHz to 6000 MHz: -10 dBm to 0 dBm, generally increasing as frequency decreases

If we calibrate input with calibrate generator, we connect the output to the input, you can also calibrate the output.
You store the value output of 1000 points.
This way you can have a calibrated output.

When you have a calibrated output, Connecting two identical antennas, una su output e una su input, It's easy to get the antenna curve and compensate for the input curve.
Antennas cost 10 euros, buy two or making two identical does not cost anything.

gioreva, 2017-04-28:
What are this parameters ?
Sample Rate. (On setup)
Bandwith. (On setup)
LNB LO: (On setup)
Bin Size: (On Main)

xmikos, 2017-04-28:
Sample rate is really basic parameter of every SDR (use Wikipedia), bandwidth is bandwidth of filters (only some SDR devices supports it), LNB LO is frequency of local oscillator if you are using upconverter or downconverter, Bin size is size of FFT bin (you can call it "resolution" if you want, e.g. if you have sample rate of 2.56 MHz and bin size 10 kHz, there will be 256 FFT bins).

xmikos, 2017-04-28:
Btw. this is issue tracker for resolving errors or for feature requests, not general discussion forum for basic informations about SDR, digital signal processing, etc.

xmikos, 2017-04-28:
As for optional calibration, I am working on feature that will allow to subtract baseline from data. Maybe this will be also usable for calibration. But there can't be some "general" predefined calibration for everyone, because every device can be somehow different and every users setup (cables, antennas, etc.) will definitely be different.

gioreva, 2017-05-01:
Can you add Output panel for setup RF output of hack RF ?
Frequency, amplitude, modulation.
And funcion for calibration output amplitude, connecting output to input.

What downloads is latest SimpleSoapy version 1.5.1 in ?
Already ave last 
QSpectrumAnalyzer-2.2.0-win-x64.msi
PothosSDR-2017.03.12-vc14-x64.exe

gioreva, 2017-05-05:
Received now HackRF one.
With CubicSDR work well, but with your nothing appears.
Start buttor returns immediately available.
On Setup I put.
Backend= hackrf_seweep
No device and no parameter

xmikos, 2017-05-05:
Please try it also with default soapy_power backend, that uses same library as CubicSDR (SoapySDR) and specify driver=hackrf as device string. I am not sure if hackrf_sweep utility (used by hackrf_sweep backend) works on Windows, but IMHO it should work too. You can also try running `hackrf_sweep.exe` from console and tell me what it outputs.

gioreva, 2017-05-05:
hackrf_sweep crash windows:
https://pastebin.com/p1jmF7yQ
Once he has done a lot of similar lines.
0 total sweeps completed, 0.00 sweeps/second

This is "soapy_power -d driver=hackrf --debug" respons
It once worked, but very slow.
From 80 to 100Mhz on some seconds.
30 to 6000 Mhz never worked.
Now not work noting.

gioreva, 2017-05-05:
This:
https://pastebin.com/R3r4cfDA

xmikos, 2017-05-05:
Did you used Zadig to install USB drivers for HackRF and then rebooted computer? I can't help you much more with hackrf_sweep (it is not my app), but soapy_power is timing out, which could be possibly caused by some USB issues.

xmikos, 2017-05-05:
Btw. did you consider using Linux for SDR work? Linux is generally much better supported (e.g. GNU Radio works much better, there are no problems with USB drivers, etc.).

gioreva, 2017-05-05:
I  can't add other PC on my laboratory.

Now hackrf_sweep work, but nothing appairs.
https://pastebin.com/YhgHYG4G

On setup:
backend= hackrf_sweep
add par=
device=
sample=20
bandw=0
LNB=1

start=88
stop=108
bin=10

gioreva, 2017-05-06:
If i reset board after each run, it work. (sopy_power)
But why do I have more trouble than signal ?
https://snag.gy/J0vRnX.jpg
Are bad the board or I something wrong?

E3V3A, 2017-08-27:
I can confirm that this doesn't work. 
I got some "output" at first, when I realized it was from the soundcard and not the RTL. Then it simply didn't work again. Did you guys actually test this on Windows using simple SDR? If you did, it would be extremely more useful to have a screen-shot of the correct `Settings`.

bicurico, 2017-11-27:
Hi,

I have exactly the same issues... Using HackRF One with Windows 10 64 bit.

First there was the PATH issue.
Then I had Python 2.7 installed.

Now that I fixed all this, I see data coming in when starting with "python -m qspectrumanalyzer", but no refresh on the screen. After some time it will crash. If i press the RESET button, it will crash right away.

Under Linux it works well.

If I run it with soapysdr, it will crash, as well. I *think* it has to do with the FFT libraries!

Has this been tested successfully by someone? It would be awesome to get this to work, as this is the only application I am missing under Windows and it would be a lot of trouble if I had to install Linux just for this.

Thanks,
Vitor

porlock, 2024-08-29:
> Hi,
> 
> I have exactly the same issues... Using HackRF One with Windows 10 64 bit.
> 
> First there was the PATH issue. Then I had Python 2.7 installed.
> 
> Now that I fixed all this, I see data coming in when starting with "python -m qspectrumanalyzer", but no refresh on the screen. After some time it will crash. If i press the RESET button, it will crash right away.
> 
> Under Linux it works well.
> 
> This is the error I get:
> 
> ```
> call hackrf_sample_rate_set(20.000 MHz)
> call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
> Sweeping from 0 MHz to 6000 MHz
> Stop with Ctrl-C
> Traceback (most recent call last):
>   File "C:\Users\Vitor\AppData\Local\Programs\Python\Python36\lib\site-packages\qspectrumanalyzer\backends\hackrf_sweep.py", line 143, in run
>     self.parse_output(buf)
>   File "C:\Users\Vitor\AppData\Local\Programs\Python\Python36\lib\site-packages\qspectrumanalyzer\backends\hackrf_sweep.py", line 114, in parse_output
>     data = np.fromstring(buf[16:], dtype='<f4')
> ValueError: string size must be a multiple of element size
> 0 total sweeps completed, 0.00 sweeps/second
> 0 total sweeps completed, 0.00 sweeps/second
> 
> Couldn't transfer any data for one second.
> 
> Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
> Total sweeps: 0 in 2.00320 seconds (0.00 sweeps/second)
> 
> C:\Users\Vitor>hackrf_stop_rx() done
> hackrf_close() done
> hackrf_exit() done
> fclose(fd) done
> exit
> ```
> 
> If I point to hackrf_sweep of the PothosSDR installation, it will get further:
> 
> ```
> hackrf_sweep params:
> {'bin_size': 1000.0,
>  'crop': 0,
>  'device': 0,
>  'gain': 20,
>  'hops': 0,
>  'interval': 0,
>  'lna_gain': 8,
>  'ppm': 0,
>  'sample_rate': 20000000.0,
>  'single_shot': False,
>  'start_freq': 100.0,
>  'stop_freq': 400.0,
>  'vga_gain': 12}
> 
> call hackrf_sample_rate_set(20.000 MHz)
> call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
> Sweeping from 100 MHz to 400 MHz
> Stop with Ctrl-C
> 27 total sweeps completed, 26.99 sweeps/second
> 53 total sweeps completed, 26.49 sweeps/second
> 80 total sweeps completed, 26.65 sweeps/second
> 108 total sweeps completed, 26.99 sweeps/second
> 135 total sweeps completed, 26.99 sweeps/second
> 162 total sweeps completed, 26.98 sweeps/second
> 189 total sweeps completed, 26.98 sweeps/second
> ```
> 
> But then it crashes anyway...
> 
> If I run it with soapysdr, it will crash, as well. I _think_ it has to do with the FFT libraries!
> 
> Has this been tested successfully by someone? It would be awesome to get this to work, as this is the only application I am missing under Windows and it would be a lot of trouble if I had to install Linux just for this.
> 
> Thanks, Vitor

I have exact same issue, this is old soft .. and i don't see any alternative :(