xmikos/qspectrumanalyzer #33: No devices found! in Windows 7

stevegoode, 2017-05-12:
Hello,
I am not a programmer.  I am trying to install qspectrumanalyzer on a Windows 7 machine.  I installed PothosSDR-2017.05.02-vc14-x64.exe and QSpectrumAnalyzer-2.2.0-win-x64.msi.  These programs generated new folders under C:\Program Files of PothosSDR and qspectrumanalyzer.  I am able to run gqrx in the PothosSDR folder and it works.   When I run qspectrumanalyzer I get No devices found! in the device setting window.  If I run soapy_power --info in the qspectrumanalyzer folder I get:

C:\Program Files\QSpectrumAnalyzer>soapy_power --info
WARNING: Incorrect RX channel number, using channel 0 instead!
Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae
ine 2, in <module>
  File "soapypower\__main__.py", line 301, in main
  File "soapypower\__main__.py", line 106, in device_info
  File "site-packages\simplesoapy.py", line 384, in get_frequency_range
IndexError: tuple index out of range
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

Can you tell me what I am doing wrong?
Thanks,

Steve

xmikos, 2017-05-15:
Hello,
it seems that you have not specified correct device driver (on Windows SoapySDR uses your sound card as default driver).

Try to run `soapy_power --detect` to see if it detects your SDR device. Then run `soapy_power --info -d driver=NAME_OF_DETECTED_DEVICE_DRIVER` to see if it works. You have to specify `driver=NAME_OF_DETECTED_DEVICE_DRIVER` also in QSpectrumAnalyzer settings (there is Device string form field for it).

stevegoode, 2017-05-15:
Hello,

When I run soapy_power --detect I get no devices found.

Steve

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

u:\>c:

C:\>Program Files
'Program' is not recognized as an internal or external command,
operable program or batch file.

C:\>cd Program Files

C:\Program Files>cd QSpectrumAnalyzer

C:\Program Files\QSpectrumAnalyzer>soapy_power --detect
Detected SoapySDR devices:
  No devices found!

xmikos, 2017-05-15:
Hmm, maybe there is some incompatibility with latest PotosSDR bundle. I will try it and release new version if this is the cause.

peeraxel, 2017-05-24:
Same problem here trying to make it work with Airspy and Win 7 / 64 bit: Soapy_power --detect shows: no devices found. QSpectrumAnalyzer does not work either. USB drivers are installed (btw after upgrading from WinUSB v1 to v6 with Zadig: SDR# still works). 
And the Airspy is detected by SoapySDRUtil (see below). So which magic parameters are missing?

C:\Program Files\PothosSDR\bin>SoapySDRUtil --probe="driver=airspy"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=airspy
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-57-g
6f87cca4

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=Airspy
  hardware=Airspy
  device_id=0
  origin=https://github.com/pothosware/SoapyAirspy

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Bias tee - Enable the 4.5v DC Bias tee to power SpyVerter / LNA / etc. vi
a antenna connection.
       [key=biastee, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=65536]
  Antennas: RX
  Full gain range: [0, 45] dB
    LNA gain range: [0, 15] dB
    MIX gain range: [0, 15] dB
    VGA gain range: [0, 15] dB
  Full freq range: [24, 1800] MHz
    RF freq range: [24, 1800] MHz
  Sample rates: 6, 3 MSps

Timmmm, 2017-06-26:
Same:

```
C:\Program Files\QSpectrumAnalyzer>soapysdrutil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-57-g6f87cca4

Found device 0
  driver = lime
  index = 0
  label = USB 3.0 (LimeSDR-USB) [USB] 9062000C42615
  media = USB
  module = STREAM
  name = USB 3.0 (LimeSDR-USB)
  serial = 0009062000C42615

C:\Program Files\QSpectrumAnalyzer>soapy_power --detect
Detected SoapySDR devices:
  No devices found!
```

I've tried `--verbose` but it just printed a message about the fftw module being found. Is there any way to debug this?

benhaha, 2017-06-28:
I clicked QSpectrumAnalyzer->settings->device(? button)
dialog shows:
  Detected SoapySDR devices:
  No devices found!

and then, i try to run "soapy_power.exe --detect" from windows cmd,
it shows same message as above.

but the SDR# works and SoapySDRUtil.exe in PothosSDR can find my device.

my software enviroment:
- windows 7 64-bit
- Python 3.6
- QSpectrumAnalyzer 2.2.0
my harware :
- hackrf one （firmware version 2017.02, 2015.* also try）

wb8vsu, 2017-07-31:
My computer is a Windows 10 system, but having the same issue. One difference is it worked on Friday, Before Friday I had loaded the drivers back in March, but only started to play with the LimeSDR Friday. Once it was working I decided it was time to update to current versions. So I updated to current versions on Saturday and it quit working. 

I decided to install on a second computer to see if I could duplicate it working on Friday. I was not able to duplicate, but it gave me a different error message. I loaded the current version of QspectrumAnalyer as it was not installed on the second computer, but suspect I must have been a older version of Qspectrumana;yzer on the computer I was using on Friday.  The messages on the new computer were similar to:

C:\Program Files\QSpectrumAnalyzer>soapy_power --info
[1m[31m[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/airspySupport.dll)
  LoadLibrary() failed: The specified module could not be found.

I looks like the delimiter between directories is not correct. As in / verse \. The first part has Windows version and the last part Linux / Unix version.

I am willing to assist in debug of the issue. I used to do programing many years ago.

Jim

Timmmm, 2017-07-31:
@wb8vsu It's probably not a delimiter issue. Windows supports `\` and `/` as directory delimiters. Either that file (airspySupport.dll) does not exist, or when it is loaded it tries to open another file that doesn't exist. Easy enough to check if airspySupport.dll exists. If it does then I think you can check which files it opens using [Logger](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/logger-and-logviewer).

In any case, I think that is an unrelated error about [Airspy](http://airspy.com/) support. I don't see how it would affect LimeSDR.

wb8vsu, 2017-07-31:
I was not aware of the change in recognition of the delimiter. Learned something new. 

I have the same error with the lime module, just did not post the many errors. Here is the LimeSDR file error:
[1m[31m[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/LMS7Support.dll)
  LoadLibrary() failed: The specified module could not be found.

I looked at the files in PothosSDR and the modules were in the following directory:
   C:\Program Files\PothosSDR\lib\SoapySDR\modules0.6
While the program errors indicate they were in:
   C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2

Do not understand why the one computer gives a better error message.

Timmmm, 2017-07-31:
I think that is a separate issue. You could make a new Github issue for it but I expect just uninstalling everything and installing from scratch would fix it (or at least get you back to the "No devices found!" error.

wb8vsu, 2017-07-31:
I suspect that is the cause of the no device error. I removed Pothos and reinstalled the version from May 2 which includes the missing directory and now it is finding USB devices.

wb8vsu, 2017-07-31:
It is now working again as it can find the LimeSDR. I am using the new USB driver for LimeSDR, verse using Zadig to change the USB driver.

Jonoro, 2017-09-05:
Another newbie here with the same problem - "no devices found"
Windows 10 with all updates
generic 820T2 with RTL2838U  USb device.
Installed Pothos (used the 2 May version as earlier it was posted that this version would detect the dongle)
gqrx.exe works fine picking up the dongle and even RF!
Qspectrumanalyser reports "No devices found"

Any and all  help appreciated very much
------------------------------------------------------------
                                  Running rtl_eeprom.exe results in:
PS C:\program files> cd pothossdr
PS C:\program files\pothossdr> cd bin
PS C:\program files\pothossdr\bin> rtl_eeprom.exe
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00000001
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
--------------------------------------------------------------------------                                 
Running soapysdrutil.exe –find results in:

PS C:\program files\pothossdr\bin> soapysdrutil.exe --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-52-g86c74ee9

Found Rafael Micro R820T tuner
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = Microphone (Realtek High Definition Audio)

Found device 2
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T
-------------------------------------------------------------------

                        Running soapysdrutil.exe --make="driver=rtlsdr" results in:

PS C:\program files\pothossdr\bin> soapysdrutil.exe --make="driver=rtlsdr"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Make device driver=rtlsdr
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-52-g86c74ee9

Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
  driver=RTLSDR
  hardware=RTLSDR
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

PS C:\program files\pothossdr\bin>
------------------------------------------------------------------------------
                                  Running soapysdrutil.exe --find again produces:

PS C:\program files\pothossdr\bin> soapysdrutil.exe --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-52-g86c74ee9

Found Rafael Micro R820T tuner
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = Microphone (Realtek High Definition Audio)

Found device 2
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T

-----------------------------------------------------------------

Checking in qspectrumanalyser for the device produces:
Detected SoapySDR devices:
No devices found!

wb8vsu, 2017-09-05:
I apologize, for not posting sooner, I was pretty sure it affected all USB devices and not just the LImeSDR. Although it will work with that version, you have to install an older version first, then update to that version. On the computer that it worked on had the Soapy 0.4.x libraries and then one of the early versions of removed the old version and replaced it with 0.5 libs. The May version installs 0.6, but does not uninstall the 0.5 libs. The newest versions seem to disable access to 0.5 libs which are required.

My install sequences was (do not know what the ones was prior to 3-12):

PothosSDR-2017.03.12-vc14-x64
Which installed: C:\Program Files\PothosSDR\lib\SoapySDR\modules0.5-2

PothosSDR-2017.05.02-vc14-x64  (works but only if a previous version installed)
Installed: C:\Program Files\PothosSDR\lib\SoapySDR\modules0.6

PothosSDR-2017.05.16-vc14-x64 (works but only if a previous version installed)
Installed: C:\Program Files\PothosSDR\lib\SoapySDR\modules0.6

Both PothosSDR-2017.06.21-vc14-x64 and PothosSDR-2017.07.27-vc14-x64 prevent QSpectrumAnalyzer from finding any USB devices including USB mice.

Jonoro, 2017-09-05:
Thank you for the very quick response.
I removed all traces of the installations then installed 
03.07, then 03.12, then 05.02, then 05.16
I then installed using the .msi the latest qspectrum analyzer
Finally ran zadig and reinstalled the driver but unfortunately I still get the same result.
qspectrumanalyzer produces this:
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-57-g6f87cca4

No devices found!

A little more verbose than before but still no SDR.

Jonoro, 2017-09-05:
I should have added that I looked and in the SoapySDR folder are 2 folders:
modules0.5-2
modules0.6

wb8vsu, 2017-09-06:
Mine is back to not finding any usb devices including mice, etc. Going to have to figure out what I did to break it.

wb8vsu, 2017-09-07:
I am new to SDR, Python and Pothos although I did some programing many years ago. So I do not really know what I am doing. I have not found what I did to break it on my system, but leaning towards the fact I had removed Python 3.5 and later installed a newer version 3.6. I currently have a 2.7 and 3.6 version of Python installed. When Qspectrumanalyzer was compiled, Python was at version 3.4. I may try installing the older version 3.4 to see what happens. 

When I ran soapy_power.exe in a cmd window I got the following error:

C:\Program Files\QSpectrumAnalyzer>soapy_power.exe --debug --detect
DEBUG: pyfftw module found (using 4 threads by default)
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966.py", line 2, in <module>
  File "soapypower\__main__.py", line 295, in main
  File "soapypower\__main__.py", line 72, in detect_devices
  File "site-packages\simplesoapy.py", line 37, in detect_devices
KeyError: 'label'
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

Jonoro, 2017-09-12:
I tried many variations earlier releases of  Pothos and portable Qspectrum but could never get it to find the SDR device.
I have virtually no programming knowledge so I took the easy way out and used a portable version of Ubuntu, and it runs first time.
So success in that at least I have qspectrumanalyzer running.

delray, 2017-09-14:
Similar issue here. Pothos and all the elements work fine (GQRX, CubicSDR GnuRadio) and SoapySDRUtil finds my device. (HackRF) But when I try and use Qspectrumanalyzer, it crashes on close no matter what backend I use. Also soapy_power does not detect any devices. 
Im on windows10 64bit, and the SoapySDR modules are 0.6.  I have this working in Ubuntu, but for a variety of reasons it would be great to get it going in windows.

wb8vsu, 2017-09-14:
I played a little today and at least on my system when it runs it is not finding some of the support files. I added where it is installed to the system environment Path, and it started to find the files. I suspect that it is not finding the Pothos Soapy files as well. I will add that location to the Path in a few minutes.

errors before changing the Path:
C:\Program Files\QSpectrumAnalyzer>soapy_power --debug --detect
DEBUG: pyfftw module found (using 4 threads by default)
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966.py", line 2, in <module>
  File "soapypower\__main__.py", line 295, in main
  File "soapypower\__main__.py", line 72, in detect_devices
  File "site-packages\simplesoapy.py", line 37, in detect_devices
KeyError: 'label'
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

After changing the Path:
C:\Program Files\QSpectrumAnalyzer>soapy_power --debug --detect
DEBUG: pyfftw module found (using 4 threads by default)
Detected SoapySDR devices:
  No devices found!

wb8vsu, 2017-09-14:
I ran soapysdrutil --info and it could not find any module drivers, where it used to work. 

So I removed Pothos and installed PothosSDR-2015.12.20-vc11-x64, , and reran soapysdrutil --info,. It had a number of issues, but found the module drivers. I then reinstalled PothosSDR-2017.03.07-vc14-x64 and ran soapysdrutil --info, which also found the module drivers without errors, although earlier today it could not. (output below).

Qspectrumanalyzer is now working, although it still can not find USB  devices in the settings. In my case it still had the old setting to the LimeSDR device so I did not need to find a new device.

I suspect it Qspectrumanalyzer needs recompiled linking to the newer version of Pothos.

C:\Program Files\QSpectrumAnalyzer>soapysdrutil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.5.5-PothosSDR-2017.03.07-vc14-x64
API Version: v0.5.2
ABI Version: v0.5-2
Install root: C:\Program Files\PothosSDR
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/airspySupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/audioSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/bladeRFSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/HackRFSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/LMS7Support.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/miriSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/osmosdrSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/RedPitaya.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/remoteSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/rtlsdrSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/s9cSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/sdrPlaySupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.5-2/uhdSupport.dll
Loading modules... Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

done
Available factories...airspy, audio, bladerf, hackrf, lime, miri, null, osmosdr, redpitaya, remote, rtlsdr, s9c, sdrplay, uhd,

wb8vsu, 2017-09-15:
I suspect there are two or more issues.  There seems to be a issue at least on my system with the SoapySDR only working if a old install is done before installing a newer version. Second Qspectrumanalyser is not able to access the SoapySDR correctly and is sensitive to which library version.

To see what would happen I removed Qspectrumanalyzer and reinstalled. I now get the following error, again:
C:\Program Files\QSpectrumAnalyzer>soapy_power --debug --detect
DEBUG: pyfftw module found (using 4 threads by default)
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966.py", line 2, in <module>
  File "soapypower\__main__.py", line 295, in main
  File "soapypower\__main__.py", line 72, in detect_devices
  File "site-packages\simplesoapy.py", line 37, in detect_devices
KeyError: 'label'
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

The file soapy_power.py is no longer there, plus I suspect it was supposed to be complied into a pyd file (dll).

Next I am going to uninstall Pothos and try installing the October version of Pothos, as I suspect that was the the first one I had installed and then updated.

wb8vsu, 2017-09-26:
I discovered that as I had installed Visual Studio with Python, that anytime a .py file tried run, it tried to open up in Visual Studio instead of having Python run it. 

I manually ran soapy_power.py and had the following error:

C:\Program Files\QSpectrumAnalyzer>C:\Users\jbach\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\qspectrumanalyzer\backends\soapy_power.py --detect --debug
  File "C:\Users\jbach\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\qspectrumanalyzer\backends\soapy_power.py", line 196
    print(e, file=sys.stderr)
                 ^
SyntaxError: invalid syntax

Qspectrumanalyzer is still running but still can not detect devices.

wb8vsu, 2017-09-26:
On my system I had to install Python 2.7 as others had not updated to Python 3 yet. For reasons I do not understand the script was running in Python 2.7 and not Python 3 as it should have. I removed Python 2.7 and the error went away. Most likely when I did my original install Python 2.7 was not installed on the system. Still will not detect devices.

wb8vsu, 2017-09-27:
Victory..  I now have it finding USB devices again. The issue seems to be Pothos and the fact it does not do a clean uninstall. What I did to clear the issue was to uninstall Pothos, then deleted its directory "C:\Program Files\PothosSDR".  Then I reinstalled the 2nd March version of Pothos and now it can find the USB devices again.  

Here is the output of running it in a cmd window (although it did not see the mouse).

C:\Program Files\QSpectrumAnalyzer>soapy_power --debug --detect
DEBUG: pyfftw module found (using 4 threads by default)
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

Detected SoapySDR devices:
  driver=audio, device_id=0, label=Default Device
  driver=audio, device_id=3, label=Microphone (USB Audio CODEC )
  driver=audio, device_id=4, label=Microphone Array (Realtek High Definition Audio(SST))
  driver=lime, serial=0009062000C41C19, label=USB 3.0 (LimeSDR-USB) [USB] 9062000C41C19

Jonoro, 2017-09-28:
Well done wb8vsu.
I might try the windows version again.
I thought I was all there with the Ubuntu version, but have just found out that the export command returns an error, hopefully the windows version export might work??
The error I get is:
ImportError: cannot import name 'QtSvg'
I'll post this as a new topic as its off-topic for the windows 7 thread.

Jonoro, 2017-09-29:
Leave a I couldn't find a 2nd March version of Pothos here:
http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D
Please could you check again which version worked for you?

wb8vsu, 2017-09-29:
I meant the second version in March.  What I have installed is: PothosSDR-2017.03.12-vc14-x64

jfitter, 2017-10-06:
Same here. Clean W7 SP1 X64 system. SoapySDRUtil works fine and the device is detected. All of the Pothos applications work fine. The only thing that refuses to work is Soapy_Power. It seems incapable of detecting my SDR device, and consequently QSpectrumAnalyzer will not work either. I have spent a lot of time reading and trying to analyse this but have now wasted enough time. I have research to get on with. Pity - looks like a good tool.

xmikos, 2017-10-06:
@jfitter I should have finally time to work on QSpectrumAnalyzer again. I will try to look at it on weekend and release new version which should hopefully work on Windows again. Sorry for your troubles.

@Jonoro Just install package `python3-pyqt5.qtsvg` and export should work.

jfitter, 2017-10-06:
OK thanks. Look forward to seeing what you come up with.

Jonoro, 2017-10-07:
I tried with Win 7 SP1 and using Pothos SDR from 12th March and it did find the device, after I manually entered the device= string into Qspectrum's settings..  Scan's worked and all looked good until I tried to export the results.  Exporting or copying resulted in the application crashing.
I next tried Win 10, as I had been trying Win 7 using a live USB installation.  This time I used a "proper" installation.  Again I used the 12th March Pothos and I tried both with the .msi and portable versions of Qspectrum.  Both worked fine until export or copying which again, like Win 7, caused it to crash.

So I am now going to retry in Ubuntu after installing the package xmikos helpfully suggested above.

jfitter, 2017-10-22:
So, is QSA going to be fixed for windows? I does not work in W7 for any exe I choose, not just soapy_power. The drivers are ok because they work just fine with other apps such as cubic sdr, pothos gui, etc. It is only QSA that will not work on W7.

Jonoro, 2017-10-22:
I installed this, it works fine out of the box
https://eartoearoak.com/software/rtlsdr-scanner

Black-Catz, 2018-02-18:
I am experiencing a similar issue in windows 10 64 bit with a hackrf one. Cubic SDR and GQRX function correctly as does the rest of the PothosSDR suite for both my rtl-sdr and hackrf one. Please let me know if I can be of further assistance with log outputs or whatever.

C:\Program Files\QSpectrumAnalyzer>soapy_power --detect
Detected SoapySDR devices:
  No devices found!

C:\Program Files\QSpectrumAnalyzer>soapy_power
WARNING: Incorrect RX channel number, using channel 0 instead!
Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966.py", line 2, in <module>
  File "soapypower\__main__.py", line 317, in main
  File "soapypower\power.py", line 38, in __init__
  File "site-packages\simplesoapy.py", line 102, in __init__
  File "site-packages\simplesoapy.py", line 185, in sample_rate
  File "site-packages\simplesoapy.py", line 14, in closest
ValueError: min() arg is an empty sequence
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

C:\Program Files\QSpectrumAnalyzer>SoapySDRUtil --probe="driver=hackrf"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=hackrf
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.003.000-0-gef157678

[INFO] Opening HackRF One #1 xxxxxxxxxxxxxxxxxx...

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=HackRF
  hardware=HackRF One
  clock source=internal
  part id=a000xxxxxxxxxxx
  serial=0000000000000000xxxxxxxxxxxxxxxxxxxx
  version=2017.02.1

wb8vsu, 2018-02-25:
Xmikos, I will have some time available in about 3 weeks. Is there anything I can do to help with resolving the problem? At the very least I am willing to test new versions.

tdbone1, 2018-11-29:
Im on windows 10 x64 fully updated to this point. (fresh install no previous versions of pothos or qspectrum)
installed "PothosSDR-2018.11.04-vc14-x64.exe | 05-Nov-2018 03:04 | 89M"
and
"QSpectrumAnalyzer-2.2.0-win-x64"

C:\Program Files\QSpectrumAnalyzer>soapy_power.exe --info
WARNING: Incorrect RX channel number, using channel 0 instead!
Traceback (most recent call last):
  File "build\temp.win-amd64-3.6\workpath\soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966.py", line 2, in <module>
  File "soapypower\__main__.py", line 301, in main
  File "soapypower\__main__.py", line 106, in device_info
  File "site-packages\simplesoapy.py", line 384, in get_frequency_range
IndexError: tuple index out of range
Failed to execute script soapy_power.py.399ce066-d3a8-43ab-a0ec-ae2654d1d966

C:\Program Files\QSpectrumAnalyzer>soapy_power --detect
Detected SoapySDR devices:
  No devices found!

C:\Program Files\QSpectrumAnalyzer>soapysdrutil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.13.1.0-1-gd3b7e90a
Found device 0
  driver = lime
  index = 0
  label = LimeSDR Mini [USB 3] 1D3BD0C4F0CDFF
  media = USB 3
  module = FT601
  name = LimeSDR Mini
  serial = 1D3BD0C4F0CDFF

C:\Program Files\QSpectrumAnalyzer>soapy_power --detect
Detected SoapySDR devices:
  No devices found!

C:\Program Files\QSpectrumAnalyzer>

everything works fine in ubuntu x64 desktop of course.

wb8vsu, 2018-12-01:
Simple LimeSDR  spectrum analyzer app is:  signal_analyzer.grc

CarneyResearch, 2020-11-08:
I am trying to use QSpectrumAnalyser and have this problem, of the app not finding devices, however I does appear to be resolved in these threads. Was it ever resolved?

CarneyResearch, 2020-11-09:
Sorry for my poor grammar on my previous posting...
I am trying to use QSpectrumAnalyser and have this problem (of the app not finding devices... in my case RTL-SDR). For some it worked right off first time. I tried the suggestion to install PothosSDR for March 12, 2017 with no luck. Was it ever resolved to get QSpectrumAnalyser to (automatically) list the SDR device in setup?

wb8vsu, 2020-11-09:
Your English is fine. 

It works for some people if you have a version of Pothos installed that has the right version of simple soapy libraries. I had to move on to newer versions of Pothos and gave up on using Qspectrumanalyzer. 

Qspectrumanalyzer does allow a custom SDR driver. Maybe some day someone can generate a custom Python driver linked to the new simple soapy libraries. Or maybe someone can create a GNU Spectrum Analyzer. 

Some of the RTL-SDR dongles have a spectrum analyzer app. 

I have used URH a couple of times, but not as nice as Qspectrumanalyzer. 

⁣Jim Bacher, WB8VSU 
wb8vsu@arrl.net 
https://trc.guru​

CarneyResearch, 2020-11-09:
Thanks Jim (WB8VSU),

Seems crazy soapySDRutil found and listed my RTL-SDR as being found but QSA can't see it..

C:\Users\CTRL_Bench>soapySDRutil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

Found Rafael Micro R820T tuner
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = Microphone (USB Audio Device)

Found device 2
  default_input = False
  default_output = False
  device_id = 3
  driver = audio
  label = Microphone (2- High Definition Audio Device)

Found device 3
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T

CarneyResearch, 2020-11-09:
Jim  (WB8VSU),

I should have mention that I also tried the suggested version of POTHOS mentioned in previous postings: No dice. Didn't work.

CarneyResearch, 2020-11-09:
Hallo XMIKOS (Michal),
Are you monitoring this chat site?

CarneyResearch, 2020-11-10:
Xmikos  & Jim (WB8VSU)

My coding days are long gone (1970's) so I have to deal with the logic of this....if the code and/or links are in error, as has been speculated in these threads, how can anyone be running QSA?

wb8vsu, 2020-11-10:
The question is am I correct that it's the simple soapy library change that impacted the application? I had looked into it a little and there was a previous Simple Soapy change he had tweaked the app for, but I could tell what actually changed. 

If I am then the following would work. 

There is an ability to select an external driver. Someone who knows Python could create a Python script to fix the issue for Linux and Windows. 

⁣

CarneyResearch, 2020-11-10:
Looking forward to a brilliant/genius/knowledgeable python coder creating this. I’ll test immediately with my current RTL-SDR.

P.S. If QSA works, I’ll replace my RTL-SDR with a hackRF One, to view the whole 1-6Ghz spectrum in a single window , and get rolling research of an ancient artifact.

CarneyResearch, 2020-11-10:
Not that it should matter here, but note that ‘Carney Research’ is NOT a business but my hobby. I am retired now and using my engineering experience and curiosity to seek answers to questions I never had time to investigate when I was a working stiff.

CarneyResearch, 2020-11-11:
I used the Windows install of QSA. Interesting note in the v2,2.0 (March 2017) release notes says:
"soapy_power dependency is now optional (it is still required in setup.py, because it is default backend, but QSpactrumAnalyzer will now actually run even without it)"

Still, QSA does NTO show my RTL_SDR device in the device line in settings. When I select the '?' to the right when my device should show I got this information copy/pasted below, It correctly shows my "Generic RTL2832U OEM :: 00000001" as do all the other major SDR based apps including SDR#:

Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

Detected SoapySDR devices:
  driver=audio, device_id=0, label=Default Device
  driver=audio, device_id=2, label=Microphone (USB Audio Device)
  driver=audio, device_id=3, label=Microphone (2- High Definition Audio Device)
  driver=rtlsdr, serial=00000001, rtl=0, label=Generic RTL2832U OEM :: 00000001

Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-0-gc705922a

Selected device: Audio
  Available RX channels:
    0
  Available antennas:
    RX
  Available tunable elements:
    RF
  Available amplification elements:
    
  Available device settings:
    sample_offset ... Stereo Sample Offset - Offset stereo samples for off-by-one audio inputs. (default: 0)
  Available stream arguments:
    chan ... Channel Setup - Input channel configuration. (default: mono_l)
  Allowed gain range [dB]:
    0.00 - 0.00
  Allowed frequency range [MHz]:
    0.00 - 6000.00
  Allowed sample rates [MHz]:
    0.00, 0.01, 0.01, 0.01, 0.01, 0.02, 0.02, 0.03, 0.04, 0.05, 0.09, 0.10, 0.18, 0.19
  Allowed bandwidths [MHz]:
    N/A

CarneyResearch, 2020-11-14:
Thanks Jim (WB8VSU), **Xmikos** and the all the rest, 
_Here is my sequence...._
1. I am on 64-bit Windows 10. I had problems with QSA seeing my RTL-SDR. 
2, I received and now using 'HackRF One' as my SDR hardware. Works fine with SDR# and SDR Angel all other SDR apps except QSA.
3. I have uninstalled QSA and Pothos (including deleting of residual directories (if any)) as suggested by Jim and others on this thread.
4. I installed Pothos 05-02 and 05-16 and have _both_ the Soapy directories 5.2 and 6.
5. ALL the Soapy inquiry commands listed in this thread now check out and correctly show 'HackRF' device.
6. QSA comes up and runs. Under 'Settings' the 'backend' is set to Soapy_Power'.  
Sounds excellent so far however....
**RESULTS**:  my QSA, although running on screen, does NOT have the 'Control' section of the right hand menu..._so I can't start_ QSA and therefore cannot determine if QSA is or is not finding my 'HackRF ONE' SDR device...or does the missing 'Control' menu indicate that it is not find the HackRF SDR?
Under 'Settings' I have tried various 'Device' setting as <blank>, HackRF and hackrf. When checking devices with Soapy it lists driver as 'hackrf'.

**Xmikos**...I need, we need, your help to resolve why QSA is sometimes not hooking in the SDR...if missing the QSA Control menu is just a result of SDR not being hooked into QSA. Ask any questions, Ready to try whatever you wish or need. I am not a programmer per se, but I can edit and tinker with your direction.

MalSpark, 2020-11-19:
Also no device found when running QSA. But rtl_power finds device and device has worked with GNU Radio companion etc. When running QSA from command line and using --debug the following shows that rtl_power is again finding device correctly but then terminates with an incorrect option later.

MalSpark, 2020-11-19:
Output is :
C:\Program Files\QSpectrumAnalyzer>qspectrumanalyzer --debug
rtl_power params:
{'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 37.0,
 'hops': 0,
 'interval': 1.0,
 'ppm': 0,
 'sample_rate': 2560000.0,
 'single_shot': False,
 'start_freq': 87.0,
 'stop_freq': 108.0}

Found 1 device(s):
  0:  Realtek, Rtl2832UDVB, SN:

Using device 0: ASUS U3100MINI_PLUS_V2
C:/Program Files/PothosSDR/bin/rtl_power.exe: invalid option -- r

MalSpark, 2020-11-19:
Using rtl_power instead of soapy as a test to debug what is going on.

MalSpark, 2020-11-19:
And when QSA configured to run soapy_power the following error when running --debug from cmd line

C:\Program Files\QSpectrumAnalyzer>qspectrumanalyzer --debug
rtl_power params:
{'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 37.0,
 'hops': 0,
 'interval': 1.0,
 'ppm': 0,
 'sample_rate': 2560000.0,
 'single_shot': False,
 'start_freq': 87.0,
 'stop_freq': 108.0}

usage: soapy_power [-h] [-f Hz|Hz:Hz] [-O FILE | --output-fd NUM]
                   [-F {rtl_power,rtl_power_fftw,soapy_power_bin}] [-q]
                   [--debug] [--detect] [--info] [--version] [-b BINS | -B Hz]
                   [-n REPEATS | -t SECONDS | -T SECONDS]
                   [-c | -u RUNS | -e SECONDS] [-d DEVICE] [-C CHANNEL]
                   [-A ANTENNA] [-r Hz] [-w Hz] [-p PPM]
                   [-g dB | -G STRING | -a] [--lnb-lo Hz]
                   [--device-settings STRING] [--force-rate]
                   [--force-bandwidth] [--tune-delay SECONDS] [--reset-stream]
                   [-o PERCENT | -k PERCENT] [-s BUFFER_SIZE]
                   [-S MAX_BUFFER_SIZE] [--even | --pow2] [--max-threads NUM]
                   [--max-queue-size NUM] [--no-pyfftw] [-l] [-R]
                   [-D {none,constant}]
                   [--fft-window {boxcar,hann,hamming,blackman,bartlett,kaiser,tukey}]
                   [--fft-window-param FLOAT] [--fft-overlap PERCENT]
soapy_power: error: unrecognized arguments: -i 1.0 0.0