xmikos/qspectrumanalyzer #41: Import error QtSvg

Jonoro, 2017-09-28:
Running Ubuntu and when I try and export the scan (Right click and export) I receive an import error.
Ubuntu is a foreign language to me, but slowly learning.

Reattached kernel driver
Traceback (most recent call last):
  File "/home/ubuntu/.local/lib/python3.5/site-packages/pyqtgraph/GraphicsScene/GraphicsScene.py", line 533, in showExportDialog
    from . import exportDialog
  File "/home/ubuntu/.local/lib/python3.5/site-packages/pyqtgraph/GraphicsScene/exportDialog.py", line 2, in <module>
    from .. import exporters as exporters
  File "/home/ubuntu/.local/lib/python3.5/site-packages/pyqtgraph/exporters/__init__.py", line 1, in <module>
    from .Exporter import Exporter
  File "/home/ubuntu/.local/lib/python3.5/site-packages/pyqtgraph/exporters/Exporter.py", line 2, in <module>
    from ..Qt import QtGui, QtCore, QtSvg
ImportError: cannot import name 'QtSvg'

xmikos, 2017-10-06:
Try to install package `python3-pyqt5.qtsvg` and let me know if it helps.

Jonoro, 2017-10-11:
Sorry for the long delay - I lost all networking! After days of trying to fix it, it's suddenly come back by itself, so I quickly installed the package you suggested.
The result is now that exporting works provided I have selected for the "entire scene"  - but the resulting image is only of the trace window, i.e. no waterfall or settings.
If instead I ask for just the "plot" to be exported then no file is created and the terminal window shows:

Traceback (most recent call last):
  File "/home/ubuntu/.local/lib/python3.5/site-packages/pyqtgraph/exporters/Exporter.py", line 77, in fileSaveFinished
    self.export(fileName=fileName, **self.fileDialog.opts)
  File "/home/ubuntu/.local/lib/python3.5/site-packages/pyqtgraph/exporters/ImageExporter.py", line 70, in export
    bg = np.empty((self.params['width'], self.params['height'], 4), dtype=np.ubyte)
TypeError: 'float' object cannot be interpreted as an integer