xmikos/qspectrumanalyzer #43: Almost 10db dips between sample stitches?

rwgast, 2017-10-21:
I don remmber ever having this problem before but depending on my gain settings I either get a dip or a peak at an interval equal to the sample rate (i.e every 2.4mhz with RTL, every 6mhz with Airspy Mini), weather it is a dip or a peak depends on gain setting.

Here is a link to an example picture with a valley/dip every 6mhz, hopefully it will make things clearer. I love this piece of software and not needing spectrum spy was what finally made me able to ditch windows.

[https://www.dropbox.com/s/rilismz9gmcgsgw/2017-10-20-222306_1366x768_scrot.png?dl=0](url)

rwgast, 2017-10-21:
Im not sure why but the link above goes to a 404... if you copy and paste the url it works just fine!

rwgast, 2017-10-22:
O btw, I forgot to mention this happens whether I use RTL_Power or Soapy_Power as the backend, so I know it has something to do with QSpectrum Analyzer itself, it does not happen in rtl_scanner or spectrum spy. I am currently running lubuntu 7.10, freshly installed. I was using Kubuntu and I did not have this problem at first and then it just popped up. I tried tweaking every optirion and removing/reinstalling QSpectrum nothing worked, so I formatted and switchedwe to lubuntu hoping to fix this problem and solve some other issues that were irking me. 

This is such a strange problem, but after formatting and installing on a fresh copy of linux I would think maybe someone else out there is having this issue, if so please let me know so I know it is not uniquely related to something specific about my system.

reldred, 2017-10-23:
You've set the crop rate right? The fact your dips are every 2.4mhz is about where the falloff is for an RTL, I'm not familiar with the Airspy personally. You don't get this problem with the hackrf_sweep backend because hackrf_sweep does the cropping in the back end.

Fire up GQRX with your RTL and watch how it tapers off at the start and end of it's 2.4mhz bandwidth. The amount it tapers off depends on what frequency range you're playing in.

Try 10% crop and see what you get.

rwgast, 2017-11-15:
So I have played with the crop setting in the QSpectrum GUI, I have not tried using the soapy_power command line switch -o (overlap) or -k (crop). Im guessing the gui sets the -k switch with the percentage entered in to the text box?

I have gotten the spectrum to look much better but now where as good as SpectrumSpy, or ear to Oak RTL Scanner. I dont know if its to high but setting crop to 65% seems to be optimal, except when I am doing a really narrow scan then I use 30%. Im still seeing all kinds of weird stuff happen at any crop rate though, it all depends on the signals received. Like noise floor show at -120db for 6mhz then will have a straight vertical rise to -110db for the next 6 mhz. And if there is a broadband signal crossing the seem between two samples I will see a mirror of the signal a few mhz away, or I may see like a .5mhz wide dip in the middle of the signal down to the noise floor. These things happen no matter what I set crop too and changing the bin size doesn't seem to have any effect.

The weird thing is the first time I used QSpectrum to sweep a filter I didn't have any of these problems, the program worked better and MUCH faster than every other program out except spectrum spy on windows, but it was just as good speed wise and much more pretty!