xmikos/qspectrumanalyzer #47: no result / display when clicking start

Hieroxym, 2017-12-05:
Hi after installing qspectrum analyzer 

when clicking on start i am not able to see any display / results.

I would like to confirm if there's anything else im required to do?

[image]

The device, USRP B210 is already being connected.

catalinalb, 2019-10-04:
Hi I have the same issue with hackrf. The sweep starts in the terminal but no display on the qspectrumanalyzer.

playaspec, 2019-10-09:
Same. Buggy with both rtl_sdr and HackRF clone running 2018.01.1 firmware and tools on OSX via Macports. I've tried both Macports Python version 3.4 and 3.7, along with py34-pyqt4, py37-pyqt5 QT bindings. hackrf_sweep is in the path, and returns the expected data when run manually. HackRF confirmed working in Gqrx.

I've tried the version through pip, from this Github repo, and the one from Mossman's github.

Clicking Start with hackrf causes the pinwheel at which point qspectrumanalyzer becomes unresponsive and frequently crashes, leaving hackrf_sweep running as an abandoned process. hackrf_sweep appears to continue running. The terminal that spawned qspectrumanalyzer is not returned until the hackrf_sweep process is killed from another terminal. It can be interrupted with ctrl-c if python hasn't already crashed, but can't be interrupted if it has.

Clicking Start with the rtl_sdr _does_ work to a degree. The waterfall (time/frequency) display updates, but nothing is displayed in the power/frequency display. Also, neither display sizes itself to represent the frequencies  in the frequency start/stop box, nor range of power levels measured.

Manually scanning with singleshot will produce one line in the waterfall, but no plot the power. Instead a single yellow line at 0db appears running the length of the frequency selected.

The one time I got a traceback, it said:

Traceback (most recent call last):
  File "/Users/playaspec/Library/Python/3.7/lib/python/site-packages/qspectrumanalyzer/data.py", line 54, in run
    result = self.task(*self.args, **self.kwargs)
  File "/Users/playaspec/Library/Python/3.7/lib/python/site-packages/qspectrumanalyzer/data.py", line 133, in update_history
    self.history.append(data["y"])
  File "/Users/playaspec/Library/Python/3.7/lib/python/site-packages/qspectrumanalyzer/data.py", line 24, in append
    self.buffer[-1] = data
ValueError: cannot copy sequence with size 1750 to array axis with dimension 1000
Abort trap: 6

jfritsch75, 2023-07-05:
Anyone ever figure out a fix for this? I've got a working hackrf_sweep instance but there is no visual display in qspectrumanalyzer, despite clear activity.

IgorShtelmakh, 2024-01-28:
> Anyone ever figure out a fix for this? I've got a working hackrf_sweep instance but there is no visual display in qspectrumanalyzer, despite clear activity.

I have similar issue. Try to right click on the chart area and select "view all". In my case it works. Seems like app has issue with automatic zoom

FroluntinoJi, 2024-08-17:
Bro, I have the same issue. Folow this, I correct this issue.
https://github.com/xmikos/qspectrumanalyzer/pull/111