xmikos/qspectrumanalyzer #66: unexpected type 'PySide2.QtWidgets.QSplitter

pauledd, 2019-01-12:
I am on gentoo linux with all dependencies installed but I get this error:

```
# .local/bin/qspectrumanalyzer --debug
Traceback (most recent call last):
  File ".local/bin/qspectrumanalyzer", line 11, in <module>
    load_entry_point('QSpectrumAnalyzer==2.2.0', 'gui_scripts', 'qspectrumanalyzer')()
  File "/home/p/.local/lib64/python3.6/site-packages/qspectrumanalyzer/__main__.py", line 535, in main
    window = QSpectrumAnalyzerMainWindow()
  File "/home/p/.local/lib64/python3.6/site-packages/qspectrumanalyzer/__main__.py", line 33, in __init__
    self.setupUi(self)
  File "/home/p/.local/lib64/python3.6/site-packages/qspectrumanalyzer/ui_qspectrumanalyzer.py", line 27, in setupUi
    self.mainPlotLayout = GraphicsLayoutWidget(self.plotSplitter)
  File "/usr/lib64/python3.6/site-packages/pyqtgraph/widgets/GraphicsLayoutWidget.py", line 26, in __init__
    GraphicsView.__init__(self, parent)
  File "/usr/lib64/python3.6/site-packages/pyqtgraph/widgets/GraphicsView.py", line 72, in __init__
    QtGui.QGraphicsView.__init__(self, parent)
TypeError: arguments did not match any overloaded call:
  QGraphicsView(parent: QWidget = None): argument 1 has unexpected type 'PySide2.QtWidgets.QSplitter'
  QGraphicsView(QGraphicsScene, parent: QWidget = None): argument 1 has unexpected type 'PySide2.QtWidgets.QSplitter'
```

tessatriolo, 2019-02-15:
I was able to work around this error by setting the QT_PREFERRED_BINDING to PyQt5:

$ export QT_PREFERRED_BINDING=PyQt5
$ python3 -c "import Qt; print(Qt.__binding__)"
PyQt5

pauledd, 2019-02-15:
Thanks, works now after I installed it with 
`sudo python3 setup.py install`
instead of
`sudo python setup.py install`

closed 2019-02-15 by pauledd