xmikos/qspectrumanalyzer #67: Tracking generator functionality

Dantali0n, 2019-02-28:
Hello, sorry if this has been asked many times before but I couldn't find any open or closed issues on this matter.

I recently tried qspectrumanalyzer with the LimeSDR and really liked it but I would like it even better if we could implement tracking generator functionality. Both the HackRF and the LimeSDR support transmission although only the LimeSDR does so in full duplex.

Adding tracking generator functionality to qspectrumanalyzer would mean it could be used to measure filters, amplifiers and all sorts of rf components their performance.

Would love to hear what you think,
All the best Dantali0n.

crazyimp, 2020-01-23:
Hi @Dantali0n , maybe the functionality you are looking for can be added to https://github.com/mihtjel/nanovna-saver/ ?

Dantali0n, 2020-01-24:
I think that it is unlikely the nanovna-saver will implement this since it specifically uses the nanovna backend and is not based on Soapy. Changing such a fundamental driver backend would basically require rewriting the entire program.