xmikos/qspectrumanalyzer #70: Question: how to use the baseline feature

sq5bpf, 2019-03-23:
I can't find any information how to use the baseline feature.

What i'd like to do:
- connect a noise source to the receiver
- save the spectrum as baseline
- use the "substract baseline" feature to get a nice flat line, preferably at 0dB
- connect a filter between the noise source and the receiver, and see the filter characteristics (and be able to see the filter passband attenuation directly on the screen)

Unfortunately i can't find any information how to save the baseline file in qspectrumanalyzer
, or what is it's format. I'm using version 2.2 (git master)
Could you help?

Patnetpatnet, 2019-03-23:
I agree. I'm finding it hard following the bit on the first page here that says "Controls should be intuitive." :) I do not understand what any of the settings are, yet would like to use it.

sq5bpf, 2019-11-24:
A note to people having the same problem:

- use the sopy_power backend

- execute qspectrumanalyser in a terminal, set the required parameters, run scan

- during execution you will see something like this in the terminal: 
Starting backend:
soapy_power -f 600.0M:1500.0M -B 100.0k -T 0.5 -d driver=rtlsdr -r 2560000.0 -p 64 -F soapy_power_bin --output-fd 29 -g 38.0 -k 37.0 -c --even --fft-window boxcar --remove-dc

- stop execution

- execute this command, but remove -c and --output-fd XX, redirect to a file:
soapy_power -f 600.0M:1500.0M -B 100.0k -T 0.5 -d driver=rtlsdr -r 2560000.0 -p 64 -F soapy_power_bin -g 38.0 -k 37.0 --even --fft-window boxcar --remove-dc
- stop execution > ~/baseline_rtl

- use the baseline_rtl file for the baseline in qspectrumanalyser

FPSychotic, 2020-09-24:
> A note to people having the same problem:
>
> * use the sopy_power backend
> * execute qspectrumanalyser in a terminal, set the required parameters, run scan
> * during execution you will see something [...]

Hi, I want to get the same, but the issue I dont know what you mean with run the program in console and run, when run the program in terminal the gui apear and I cannot write anymore in the terminal, not run command, or stop or save to file. I cannot find anywhere what is run in console. Please could you be more specific for a beginner as me? If you have a hackrf would be great id you can share your setup, I dont know what means here bin, bandwidth,rate...no idea, hackrf can sweep to 8Ghz, but only in sweep, bandwidth is 20mhz in specs, which I should use, really this software is impossible to understand for me as beginner. Would be great if you can help me a little

wernight, 2022-11-07:
Is there a way to get it to work also for hack_rf sweep? I tried the same method but it doesn't seem to have an effect.