xmikos/qspectrumanalyzer #79: issues about the limesdr's calibration setting

juzhang74, 2020-03-26:
hi,
   i meet some strange manner with qspectrumanalyzer and soapy-power backend.
   at first, this program's frequency-tuning procedure is very slow, with about 0.3s of one reset-stream() call. if we enable the cacheCalibrations switch to true, then the tuning time is reduced to 0.01s. but, sadlly news is the  qspectrumanalyzer's output will keep-no-change(same as before) until about 60s later. what's the matter i met?
   thanks.

tdbone1, 2020-11-20:
i use limesdr mini with soapy-power and also have same problems.

juzhang74, 2020-11-20:
I've solved it. But I forgot about how? give me some time to recall it

tdbone1, 2020-11-22:
> I've solved it. But I forgot about how? give me some time to recall it
ok