xmikos/qspectrumanalyzer #84: hackrf_start_rx_sweep() failed: feature not supported by installed firmware (-1005)

sl0sh, 2020-09-20:
Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .
hackrf_sweep params:
{'bin_size': 1000.0,
 'crop': 0,
 'device': 0,
 'gain': 40,
 'hops': 0,
 'interval': 0,
 'lna_gain': 16,
 'ppm': 0,
 'sample_rate': 20000000.0,
 'single_shot': False,
 'start_freq': 2.0,
 'stop_freq': 6002.0,
 'vga_gain': 24}

call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2 MHz to 6002 MHz
hackrf_start_rx_sweep() failed: feature not supported by installed firmware (-1005)
Usage:
	[-h] # this help
	[-d serial_number] # Serial number of desired HackRF
	[-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
	[-f freq_min:freq_max] # minimum and maximum frequencies in MHz
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-n num_samples] # Number of samples per frequency, 8192-4294967296
	[-w bin_width] # FFT bin width (frequency resolution) in Hz
	[-1] # one shot mode
	[-N num_sweeps] # Number of sweeps to perform
	[-B] # binary output
	[-I] # binary inverse FFT output
	-r filename # output file

Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .

I get this when I run qspectrumanalyzer.  Have tried older firmware same result.

themainframe, 2020-10-05:
Pretty much means that your HackRF One's firmware doesn't match the API used by `hackrf_sweep`. Try downloading a release of `mossmann/hackrf` (https://github.com/mossmann/hackrf/releases/tag/v2018.01.1 say), building the `host` tools and putting them in your `$PATH` and flashing the firmware image that came with that release onto your HackRF.

sl0sh, 2020-10-06:
now it says: hackrf_start_rx_sweep() pipe error (-1000)

CarneyResearch, 2020-11-15:
I am getting not able to get QSA to grab my SDR (HackRF One) either by using soapy or by hackrf_sweep settings.
Here is my sequence....
1. I am on 64-bit Windows 10. 
2, I received and now using 'HackRF One' as my SDR hardware. Works fine with SDR# and SDR Angel all other SDR apps except QSA.
3. I have uninstalled QSA and Pothos (including deleting of residual directories (if any)) as suggested by Jim and others on this thread.
4. I installed Pothos 05-02 and 05-16 and have both the Soapy directories 5.2 and 6.
5. ALL the various Soapy inquiry commands now check out and correctly show 'HackRF' device. Example below:

C:\Program Files\QSpectrumAnalyzer>SoapySDRUtil --probe="driver=hackrf"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=hackrf
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.001.001-57-g6f87cca4

[INFO] Opening HackRF One #0 17c467dc299e5ec3...

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=HackRF
  hardware=HackRF One
  clock source=internal
  part id=a000cb3c005d4f54
  serial=000000000000000017c467dc299e5ec3
  version=2018.01.1
QSA comes up and runs. Under 'Settings' the 'backend' is set to Soapy_Power'.

**_OK so sounds excellent so far however....however..._**
**RESULTS**: my QSA, although running on screen, does **NOT** have the 'Control' section of the right hand menu...so I can't start QSA and therefore cannot determine if QSA is or is not finding my 'HackRF ONE' SDR device...or does the missing 'Control' menu indicate that it is not finding the HackRF SDR?
Under 'Settings' I have tried various 'Device' settings such as , HackRF, hackrf, HackRF One #0 17c467dc299e5ec3. 

**Xmikos**...I need, we need, your help to resolve why QSA is sometimes not hooking in the SDR...if missing the QSA Control menu is just a result of SDR not being hooked into QSA. Ask any questions, Ready to try whatever you wish or need. I am not a programmer per se, but I can edit and tinker with your direction.

boehrsmarco, 2020-11-30:
Same error on my side with Ubuntu 20.04 and HackRF Firmware 2018.01.1. It works on Kali current version with the same HackRF One.

Output hackrf_info on Kali:

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc3253b707
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004c4f67

Output on Ubuntu:

hackrf_info version: git-52851ee
libhackrf version: git-52851ee (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc3253b707
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004c4f67

I installed both hackrf tools via apt.

QSpectrumanalyzer via git and pip3 install from https://github.com/xmikos/qspectrumanalyzer

stman, 2020-12-03:
I had the same error, here is the problem : You have installed an old library for "libhackrf0" that doesn't handle the sweep mode.

What you need to do is : 

- Remove the current libhackrf0 library, uninstall it with `sudo apt-get purge libhackrf0`

- Download then compile yourself the "host files" from the 2018 version of the source code, and install this freshly compiled libhackrf0 library, normally it should work perfectly.
- Eventually, if your hackrf has an old firmware not supporting the sweep mode, you will have to follow the process to upgrade the firmware of your hackrf 1, both the FPGA bitfile and the USB firmware.

In the end, in my case, upgrading the hackrf firmware was useless because I had already a sweep compatible version, but hey, it's always better to have the latest version of the firmware anyway, so in my case again, the only issue was that I had installed, with GQRX and QSpectrumanalyzer package installers, an old version of the libhackrf0 not handling the hackrf sweep mode.

Kind regards,

Stman.

Crypto-Anarchist Federation
@stman@mastodon.social

boehrsmarco, 2020-12-03:
Thank you stman! Rebuilding the libhackrf0 solved it!! Now QSpectrumanalyzer works fantastic!

BR
Marco