xmikos/qspectrumanalyzer #90: Sweeps with an Ettus B200 USRP take an eternity

aaronwoodcox, 2021-01-29:
As a user, I would like to be able to sweep frequencies using an Ettus B200 USRP at a performance level similar to what can be done with RTL HW using the rtl_power_fftw.  Sweeps with the B200 USRP are incredibly slow, for example taking upwards of 20 seconds to cover 20 MHz or less.  Where the USRP is capable of sweeping frequency ranges outside the capabilities of an RTL, it would be very desirable to be able to achieve similar performance to the RTL with the USRP.  Not quite sure why the USRP tuning would be so slow - **Is there any chance the USRP FW image is being reloaded between each tune?**  And if so, can this be rectified?

(Note, I can certainly provide more details and logs, but before going to the trouble to do so wanted to see if this is already a known issue that is being worked and/or if there might be a workaround I am unaware of that would speedup USRP captures.)

puchol, 2021-11-19:
I see the same, have not managed to cure it. I also see the same signal pattern no matter where in the spectrum I look.

aaronwoodcox, 2021-11-19:
I've all but given up on qspectrum analyzer w/ the B200...  I would love to get it working, but doesn't appear this repo is active any longer (at least no one seems to be responsive...).  One thing you might check - at least got things working slightly better for me - was to confirm that under "Settings" in the "Device" field you have "driver=uhd" and also play around a bit with the sample rate settings.  Also, if you run qspectrumanalyzer from the command line, you can at least see what it's doing under the hood (i.e., you can see the actual soapy_power command it issues, see it loading the USRP FW, etc.).  Here's an example (and note you can copy/paste and run the soapy_power command directly if you'd like to experiment with things - just be sure to remove the --output-fd <fdNum> portion of the command):
```
$ qspectrumanalyzer 
Starting backend:
soapy_power -f 430.0M:440.0M -B 1.0k -T 0.01 -d driver=uhd -r 3840000.0 -p 0 -F soapy_power_bin --output-fd 29 -w 3840000.0 -g 15.0 -k 20.0 -c --even --fft-window boxcar --remove-dc

[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-release
[INFO] [B200] Loading firmware image: /usr/share/uhd/images/usrp_b200_fw.hex...
[INFO] [b200_impl.cpp:386] [B200] Detected Device: B200
```

Anyway... hope this helps?  And if you figure anything out to get sweeps to run at anything approaching a usable rate, please let me know!!

FWIW, here's an alternative, really simple approach to getting a workable spectrum analyzer running with a B200: 
https://wiki.gnuradio.org/index.php/Guided_Tutorial_Hardware_Considerations#Creating_a_Software_Radio_Spectrum_Analyzer

puchol, 2021-11-19:
Thank you for your feedback! I was struggling with the "Device" field, nothing I tried seemed to work, I will give your format a go. Will feed back if I get anywhere useful.

NustyFrozen, 2024-09-05:
hi, i have b210 (same series) i was having the same issue try lowering the number of repeats in soapy_power it will average less (for welching) so it will look more noisy but it will dramatically decrease the sweep time since it will sweep lees i was able to get it to sweep 500mhz span/sec (with hops)

You can also try increasing sample rate