xmikos/qspectrumanalyzer #99: qspectrumanalyzer stops in 5.8GHz range after 30 seconds or so

lowtowerjones, 2022-04-11:
I'm on dragonos.  Software version is 2.1.0.  Did patch the OS and reboot, same issue.

5.13.0-35-generic #40~20.04.1-Ubuntu SMP Mon Mar 7 09:18:32 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Need to watch high spectrum to try to find a problematic device.  Not even sure it's possible due to FHSS.  But when I can in other spectrums the scan will keep going.  After scanning in 5.8 for a half minute it goes from start being greyed out to start being clickable.  the top graph stops change and I stop getting updates in the history graph on the bottom.

Using HackRF One.  Haven't tried reproducing it on another laptop/receiver.  Give me a few hours and I'll do that.

Thanks,
Low

lowtowerjones, 2022-04-11:
I walked away for a bit and it threw a semi readable gui message.  See attached. 
[image]

alphafox02, 2022-04-21:
I’ll try here locally with the same setup. Hackrf firmware I assume is 2021+ based?

alphafox02, 2022-04-21:
Okay I can’t reproduce this but here’s some things to check. Click file, settings, and ensure backend is set to hackrf sweep and not soapy power which is the default. Also make sure you have the 2021 or higher hackrf firmware flashed on the hacrkrf. I’ve tried your exact settings and it’s been running 30mins or so now.

lowtowerjones, 2022-04-21:
I'm sorry, I should have updated the ticket.  I believe the laptop I was
using it on had problems.  I couldn't reproduce it with the same receiver
on a different laptop.  And the laptop began to exhibit other issues.

alphafox02, 2022-04-21:
No problem, I actually learned about this in the process so now I’m trying out both options https://github.com/avian2/hackrf-spectrum-analyzer