xmikos/qspectrumanalyzer #102: Can't run under Ubuntu 22.04

dwarning, 2022-08-22:
I tried to install the program under the new Ubuntu release 22.04. I have Qt4 and Qt5 installed.
But got runtime error:
Traceback (most recent call last):
  File "/home/dietmar/.local/bin/qspectrumanalyzer", line 5, in <module>
    from qspectrumanalyzer.__main__ import main
  File "/home/dietmar/.local/lib/python3.10/site-packages/qspectrumanalyzer/__main__.py", line 5, in <module>
    from Qt import QtCore, QtGui, QtWidgets, __binding__
  File "/usr/local/lib/python3.10/dist-packages/Qt.py", line 1925, in <module>
    _install()
  File "/usr/local/lib/python3.10/dist-packages/Qt.py", line 1903, in _install
    our_submodule = getattr(Qt, name)
AttributeError: partially initialized module 'Qt' has no attribute 'QtGui' (most likely due to a circular import)
Under 20.04 everything was well.

dwarning, 2022-12-05:
Thanks for response.
I removed Qt.py but then the import can't work:
```
from Qt import QtCore, QtGui, QtWidgets, __binding__
ModuleNotFoundError: No module named 'Qt'
```
I have python3-pyqt5 installed. It is a nightmare with the qt installation.

courtarro, 2022-12-05:
**Actually I was mistaken**. `Qt.py` is indeed used by this project, so you need it installed after all. Reinstall it but try using `pip install --user Qt.py` and see if that works better.

Run `python3 -c "import sys; print('\n'.join(sys.path))"` to see what library paths are being searched by your Python instance.

Check `/usr/local/lib/python3.10/dist-packages/` for any lingering references to Qt that shouldn't be there.