xmikos/qspectrumanalyzer #103: Problem to show waterfall data on Kali Linux

QuixoteSystems, 2022-10-16:
All seems running fine on Kali Linux (Debian 5.19.11-1kali2 (2022-10-10) x86_64 GNU/Linux)  but before get the waterfall workingg I get this error:

```
Traceback (most recent call last):
  File "/home/kali/.local/lib/python3.10/site-packages/qspectrumanalyzer/plot.py", line 289, in update_plot
    self.waterfallImg.scale((data_storage.x[-1] - data_storage.x[0]) / len(data_storage.x), 1)
TypeError: ImageItem.scale() takes no arguments (2 given)
```

If I can change the function call in line 289 with no parameters then I don't get any error but the waterfall is not working well. My simple test:

```
#self.waterfallImg.scale((data_storage.x[-1] - data_storage.x[0]) / len(data_storage.x), 1) # line 289
self.waterfallImg.scale()

```

ianmorti, 2023-02-05:
scaling still required for me.  I've found the following modification to apply a transform works.

in plot.py, import `QtGui` (e.g. `from Qt import QtCore, QtGui`) then create and apply transform in `update_plot` function:

```python
        # Create waterfall image on first run
        if self.counter == 1:
            self.waterfallImg = pg.ImageItem()
            tr = QtGui.QTransform()
            scale_x = (data_storage.x[-1] - data_storage.x[0]) / len(data_storage.x)
            tr.scale(scale_x, 1)
            #self.waterfallImg.scale((data_storage.x[-1] - data_storage.x[0]) / len(data_storage.x), 1)
            self.waterfallImg.setTransform(tr)
            self.plot.clear()
            self.plot.addItem(self.waterfallImg)

```

betzburger, 2023-02-24:
@ianmorti 
This was a great help to me. I use macOS Ventura on an M1 Mac (Yes, qspectrumanalyzer runs on the Mac, too) and had the same issue with "no waterfall data". I tried your patch and no everything is fine!
Thank you!

fobdy, 2023-07-03:
> scaling still required for me. I've found the following modification to apply a transform works.
>
> in plot.py, import `QtGui` (e.g. `from Qt import QtCore, QtGui`) then create and apply transform in [...]

@ianmorti Maybe it makes sense to create PR? :)

jfritsch75, 2023-07-17:
> scaling still required for me. I've found the following modification to apply a transform works.
>
> in plot.py, import `QtGui` (e.g. `from Qt import QtCore, QtGui`) then create and apply transform in [...]

That patch did the trick! This issue has been driving me crazy (MBP M1 Max) - - thank you so much!  Hoping to see it get added in a future revision.

jotamjr, 2024-04-14:
> scaling still required for me. I've found the following modification to apply a transform works.
>
> in plot.py, import `QtGui` (e.g. `from Qt import QtCore, QtGui`) then create and apply transform in [...]

can confirm this works <3

justin-petermann, 2025-01-01:
Hello,

better results with that:

```
-             self.waterfallImg.scale((data_storage.x[-1] - data_storage.x[0]) / len(data_storage.x), 1)
+            self.waterfallImg.scale()
```

Best regards,
Justin.