xmikos/qspectrumanalyzer #108: It's broke . . . or so it seems?

markfsanderson, 2023-04-16:
* On the latest version of Kali Linux running on a Del 7050 I7 w/32G memory and 1g NVME drive with hackrf info:
```
└─# hackrf_info                                                              
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc273c18c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006f4362
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
```
This does appear to be the latest firmware . . .

* debug output from the command line
```
StandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 1399, resource id: 11404949, major code: 40 (TranslateCoords), minor code: 0
soapy_power params:
{'bandwidth': 1000000.0,
 'bin_size': 10.0,
 'crop': 0.0,
 'device': '/dev/hackrf-one-1-1',
 'gain': 0,
 'hops': 0,
 'interval': 10.0,
 'ppm': 0,
 'sample_rate': 1000000.0,
 'single_shot': False,
 'start_freq': 87.0,
 'stop_freq': 108.0}

[INFO] [UHD] linux; GNU C++ version 12.2.0; Boost_107400; UHD_4.3.0.0+ds1-4+b2

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

[ERROR] avahi_service_browser_new() failed: Bad state

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

WARNING: Sample rate 1000000.0 Hz is not supported, setting it to 192000.0 Hz!
WARNING: Device does not support setting filter bandwidth!
INFO: Using device: Audio
[INFO] Using format CF32.
RtAudio alsa: running realtime scheduling
INFO: repeats: 873
INFO: samples: 17460 (time: 0.09094 s)
INFO: max_buffer_size (samples): 13107200 (repeats: 655360.00, time: 68.26667 s)
INFO: buffer_size (samples): 20480 (repeats: 1024.00, time: 0.10667 s)
INFO: buffer_repeats: 1
INFO: overlap: 0.00000
INFO: bin_size: 9600.00 Hz
INFO: bins: 20
INFO: bins (after crop): 20
INFO: sample_rate: 0.192 MHz
INFO: sample_rate (after crop): 0.192 MHz
INFO: freq_range: 21.000 MHz
INFO: hopping: YES
INFO: hop_size: 0.192 MHz
INFO: hops: 110
INFO: min_center_freq: 87.096 MHz
INFO: max_center_freq: 108.024 MHz
INFO: min_freq (after crop): 87.000 MHz
INFO: max_freq (after crop): 108.120 MHz
Traceback (most recent call last):
  File "/root/.local/lib/python3.11/site-packages/qspectrumanalyzer/plot.py", line 289, in update_plot
    self.waterfallImg.scale((data_storage.x[-1] - data_storage.x[0]) / len(data_storage.x), 1)
TypeError: scale(self): too many arguments
```
1. Running from the root account, there shouldn't be any permission issues
2. No bandwidth adjustment for a spectrum analyzer?  I guess I'm missing something . . . is it fixed or?   
3. If I had the time I'd run pylint over it and see what's breaking.  but the traceback does appear to indicate a function prototype mismatch of some kind?

Thanks!

fobdy, 2023-06-10:
Doesn't work for me as well for both devices Hack RF & Pluto. I'm using Ubuntu 23

fobdy, 2023-06-10:
Due some reason it tries to call the `scale` method with the arguments but it doesn't accept any argument. Maybe the reason is that's incompatible version of `pyqtgraph` library is used.

[image]

philoteer, 2023-06-28:
Seems like the same issue: https://github.com/xmikos/qspectrumanalyzer/issues/103